project-chip/connectedhomeip #19942: I downloaded the code repository, and the compilation will report a PhaseScriptExecution error. How to solve this problem. Xcode13, iOS15

nekteckIOS, 2022-06-24:
+ gn --root=/Users/yuanrui_ios/Downloads/connectedhomeip-master gen --check out '--args=default_configs_cosmetic=[] chip_crypto="mbedtls" chip_build_tools=false chip_build_tests=false chip_disable_platform_kvs=true target_cpu="arm64" target_defines=["CHIP_ADDRESS_RESOLVE_IMPL_INCLUDE_HEADER=<lib/address_resolve/AddressResolve_DefaultImpl.h>","DEBUG=1","CHIP_ADDRESS_RESOLVE_IMPL_INCLUDE_HEADER=<lib/address_resolve/AddressResolve_DefaultImpl.h>"] target_cflags=["-target","arm64-apple-ios15.0","-arch","arm64","-flto"] target_os="ios" import("//config/ios/args.gni")'
./chip_xcode_build_connector.sh: line 160: gn: command not found
Command PhaseScriptExecution failed with a nonzero exit code

nekteckIOS, 2022-06-27:
> @nekteckIOS您使用的是哪个确切的 SHA？
> 
> 您是否有机会在您的环境中定义了 CHIP_NO_ACTIVATE？

Hello, after I downloaded it from the master branch, I modified the team id according to the readme, and did not make other changes.

closed 2022-07-01 by nekteckIOS

Jozott00, 2022-07-21:
@nekteckIOS How did you solved it?