project-chip/connectedhomeip #19944: [SQA] DUT throws the Failure Response as Can not decode Data

Saravana-kr22, 2022-06-24:
### Problem
In Thread Network Diagnostics Cluster, While reading the Security policy and Operational dataset Attribute are fails with the response as the Can not decode Data.

**Expected behavior** :  Value of attribute.
 **Actual behavior** : Can not decode Data
**Steps to reproduce** :  `./chip-tool threadnetworkdiagnostics read security-policy 83 0`,         
        ` ./chip-tool threadnetworkdiagnostics read operational-dataset-components 83 0`
**System configuration:** 
Platform: nRF52840 DK
App :  all clusters app
commit used: c1e1d018820c0fdd51f124f607b9207f31fae2ff

**Log:**

> [1656058651.991994][4823:4828] CHIP:TOO: Endpoint: 0 Cluster: 0x0000_0035 Attribute 0x0000_003D DataVersion: 1950252754
[1656058651.992113][4823:4828] CHIP:TOO: Response Failure: Can not decode Data
[1656058651.992290][4823:4828] CHIP:EM: Sending Standalone Ack for MessageCounter:133592987 on exchange 29504i

[Thread error.txt](https://github.com/project-chip/connectedhomeip/files/8974717/Thread.error.txt)

bzbarsky-apple, 2022-06-24:
That looks like it's using a chip-tool from before #18920 and a server from after that PR, so they disagree on the data type these attributes are supposed to be

@Saravana-kr22 Please double-check that chip-tool and the server are built from the same exact SHA?

Saravana-kr22, 2022-06-27:
In the latest commit c38e9159d0500a880b9c789bf5d87fa16c039633 getting the excepted outcome.

closed 2022-06-27 by Saravana-kr22