project-chip/connectedhomeip #19982: Build guide recommends Ubuntu 21.04 for Raspberry Pi, however it is end of life and not supported

presto8, 2022-06-25:
#### Problem
The building guide at https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/BUILDING.md says to install Ubuntu 21.04 using rpi-imager:

`Using rpi-imager, install the Ubuntu 21.04 64-bit server OS for arm64 architectures on a micro SD card. This release will have bluez 5.55 or newer which is required for BLE functionality.`

There are two issues with this: First, Ubuntu 21.04 is end of life as of January 2022 and no longer supported by Ubuntu (or receiving updates). Second, rpi-imager does not offer 21.04 as an option to install. 

 #### Proposed Solution

Because bluez 5.55 is needed, which is only supported >= 21.04, then the next suitable release is 22.04. However, 22.04 uses Python 3.10 which does not work out of the box with bootstrapping script.

There is a discussion over how to fix this at https://github.com/project-chip/connectedhomeip/issues/16664#

A solution is to install python 3.9 as recommend here https://github.com/project-chip/connectedhomeip/issues/16664#issuecomment-1138680477

One solution is to upgrade the building guide and copy over these additional steps. The other solution is to wait until the install script is updated to allow Python 3.10

mikaelhm, 2022-07-29:
This can be closed, verified no longer an issue. Ubuntu 22.04 and Python 3.10 is support now.

franck-apple, 2022-07-29:
Closing per above comment.

closed 2022-07-29 by franck-apple