project-chip/connectedhomeip #19989: #13 [pre-SVE] TC-DRLK-2.2 UserCodeTemporaryDisableTime attribute is not triggered

AlvinHsiao, 2022-06-27:
Problem

expected behavior vs actual behavior

At step14 after step13, DUT should send failure response to the TH, but sent success response instead.

system configuration :
Chip-Tool: RPI
DUT used : Infineon CYW30739
app use: Lock-app
chip-tool and DUT commit id : https://github.com/project-chip/connectedhomeip/commit/1e3f12041013dd71a53d035dbf2f6d142fc06c4e (pre-sve SHA)

logs
[TC-DRLK-2.2.txt](https://github.com/project-chip/connectedhomeip/files/8988573/TC-DRLK-2.2.txt)

bzbarsky-apple, 2022-06-27:
door-lock-server has:
```
    // TODO: check whether UserCodeTemporaryDisableTime expired or not.
```
and
```
    // TODO: If limit is exceeded, lock remote operations for UserCodeTemporaryDisableTime.
```
so looks like this is not implemented yet....

woody-apple, 2022-06-29:
Cert Blocker Review: @Morozov-5F are you able to take a look at this lock cert blocker? Thanks!

woody-apple, 2022-07-29:
Cert Blocker Review: Marking this as 'needs provisional' unless we have a resolution to this issue today.

PR #21408 (2022-08-01): Implement lockout for door lock
#### Problem
* Mandatory attributes UserCodeTemporaryDisableTime and WrongCodeEntryLimit are not accounted when locking/unlocking.
* Fixes #19989
* Fixes #18678

#### Change overview
Now door lock cluster tracks the lockout per each endpoint.

#### Testing
* Manual testing: attempt to lock/unlock with valid PIN, then 3 times with invalid and verified that for 10 seconds my commands were ignored (failure response and absence of LockOperationError event).
PR #21498 (2022-08-01): Implement lockout for door lock
#### Problem
* Mandatory attributes UserCodeTemporaryDisableTime and WrongCodeEntryLimit are not accounted when locking/unlocking.
* Fixes #19989
* Fixes #18678

#### Change overview
Now door lock cluster tracks the lockout per each endpoint.

#### Testing
* Manual testing: attempt to lock/unlock with valid PIN, then 3 times with invalid and verified that for 10 seconds my commands were ignored (failure response and absence of LockOperationError event).
closed 2022-08-01 by woody-apple