project-chip/connectedhomeip #19991: Door-lock Set User Command is returning Failure instead of  OCCUPIED.

Ashwinigrl, 2022-06-27:
Problem - when we try to send the  Set User command with 
[set user.txt](https://github.com/project-chip/connectedhomeip/files/8989648/set.user.txt)
[set user.txt](https://github.com/project-chip/connectedhomeip/files/8989661/set.user.txt)
repeated user index value we are getting Failure response

Actual Behavior -After giving repeated user index value we are getting Failure response

Expected Behavior - After giving repeated user index value we have to get OCCUPIED

App used - lock app
Platform - Chip-tool - RPI-4, 8GB RAM
DUT - RPI-4, 8GB RAM
Network - Ble-wifi
Commit - 2ac3ee761232b466db70c196986e90081f2c4ec3

bzbarsky-apple, 2022-06-27:
@Morozov-5F

woody-apple, 2022-07-06:
SDK Review: @Morozov-5F are you able to take a look at this?

bzbarsky-apple, 2022-07-28:
>    [1656314035.358615][2757:2762] CHIP:TOO: Error: IM Error 0x00000501: General error: 0x03 (OCCUPIED)

Note, by the way: this is _not_ the right expected value.  The rigtht expected value is a cluster-specific error, not a general error.

woody-apple, 2022-07-29:
Cert Blocker Review: Marking this as 'needs provisional' unless we have a resolution to this issue today.

PR #21281 (2022-07-29): Fix error code for the Set User command
#### Problem
* Fixes #19991.

#### Change overview
* Fixed error code sent by the SDK to Set User command.
* Changed the YAML `DL_UsersAndCredentials` test. 

#### Testing
* Verified that the door lock sends the Occupied response when Set User command is issues multiple times to the same slot.
PR #21455 (2022-07-30): Fix error code for the Set User command
#### Problem
* Fixes #19991.

#### Change overview
* Fixed error code sent by the SDK to Set User command.
* Changed the YAML `DL_UsersAndCredentials` test. 

#### Testing
* Verified that the door lock sends the Occupied response when Set User command is issues multiple times to the same slot.
closed 2022-07-29 by woody-apple