project-chip/connectedhomeip #20002: No such file or directory: '/opt/homebrew/lib/python3.9/lib2to3'

EscJump, 2022-06-27:
WELCOME TO...

           ░▓░
           ▓█▓
           ▓█▓                                                     ▒█     ▒█
      ▒██▒▒▓██▒███▒                ░▒▓▒░  ░▒▓▒░       ░░▓█▒░ ░█  █████████████░    ░▒█▒░       ░░▒░
       ░▓█████▓██░               ▒█▒░░▒▓██▓▒░░▒█▒   ░█▓▒░░▒████    █▓░    █▓░    ▒█▒░░░▒█▓░   ██▒░░
    ▒█▒░         ░██▒           ░█░     █▓     ░█░ ░▓▒      ░▓█    █▓     █▓    ▒█░░    ░██  ▒▓
    ░▓██▓░     ░██▓█░           ░█      ▓█      █░ ░█░       ██    █▓     █▓    ▓██████████  ▒█
     ░▓███▒   ▒███▒░            ░█      ▓█      █░  ██░    ░███    █▓     █▓    ░█▒░         ▒█
 ░▒████████░ ░███▓▓█▓▓▒         ░█      ▒▒      █░   ░█▓██▓█░▒█    ░▓▓█░  ░▓▓█░   ▒▓▓██▓█░   ▒▓
 ░██▒░  ▒██▒ ▒██░  ░▒█▓
         ░▓░ ░▓░

  BOOTSTRAP! Bootstrap may take a few minutes; please be patient.

Downloading and installing packages into local source directory:

  Setting up CIPD package manager...done (3.7s)
  Setting up Python environment.....Traceback (most recent call last):
  File "/Users/wyz/work/chip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 779, in <module>
    sys.exit(main())
  File "/Users/wyz/work/chip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 771, in main
    return EnvSetup(**vars(parse())).setup()
  File "/Users/wyz/work/chip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 448, in setup
    result = step(spin)
  File "/Users/wyz/work/chip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 593, in virtualenv
    if not virtualenv_setup.install(
  File "/Users/wyz/work/chip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 197, in install
    _check_python_install_permissions(python)
  File "/Users/wyz/work/chip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 145, in _check_python_install_permissions
    for file_path in os.listdir(lib2to3_path)
FileNotFoundError: [Errno 2] No such file or directory: '/opt/homebrew/lib/python3.9/lib2to3'

bzbarsky-apple, 2022-06-27:
@EscJump Do you have `lib2to3` installed with whatever Python you are using here?  If not, sounds like you should install it....

EscJump, 2022-06-27:
> @EscJump Do you have `lib2to3` installed with whatever Python you are using here? If not, sounds like you should install it....

I run brew install python3, but I can not found lib2to3. What should I do?

bzbarsky-apple, 2022-06-27:
Probably use `pip install`?

closed 2022-06-28 by EscJump