project-chip/connectedhomeip #20024: Commissionee device 1 to commissionee device 2 demo

wxiangcheng, 2022-06-28:
Hi, I want to show customer a demo that could let device 1 to tx/rx data to device 2 under the same fabric. But these 2 devices are both added by android phone as controller(commissioner). Let's assume device 1 is sweeper robotics and device 2 is just one smart bulb. When device1 get data from device 2 that it is off, and at same time device 1 knows that the time is 20:00 so device 1 sends one command to control device 2 bulb on. All commands are without phone (controller) - it is just end device to end device.

Should I write this demo totally by myself or is there any similar function that I can do in existed matter SDK demo folder?
Thanks.

wxiangcheng, 2022-06-29:
one workaround is: can one device acts as both controller and end node in matter?

hiteshmathu, 2022-07-21:
I m also looking for the same i.e Matter node to node data send. Have find any solution?

closed 2023-02-04 by stale[bot]