project-chip/connectedhomeip #20053: CommandDataIB encoding does not match spec

bzbarsky-apple, 2022-06-28:
#### Problem
The spec says, in "10.5.12.1. CommandFields":

> If there are no arguments in the Request or Response, this tag SHALL be omitted entirely.

But our SDK will absolutely encode an empty CommandFields struct in various cases.  `CommandSender::AddRequestDataInternal` will call `Encode` on the passed-in struct, and something like `OnOff::Commands::Off::Encode` will encode an empty struct for that tag.

 #### Proposed Solution
Either fix the SDK to follow the spec (sounds pretty painful) or change the spec to MAY instead of SHALL, so encoding an empty struct there is actually valid.

@yunhanw-google @mrjerryjohns @erjiaqing

bzbarsky-apple, 2022-06-28:
Filed https://github.com/CHIP-Specifications/connectedhomeip-spec/issues/5423 on the spec.

andy31415, 2022-07-11:
Spec issue marked completed.

closed 2022-07-11 by andy31415