project-chip/connectedhomeip #20093: Attributes on dynamic endpoints can't be non-volatile

wqx6, 2022-06-29:
#### Problem
Attributes on dynamic endpoints have a mask `EXTERNAL_STORAGE`. They can't be non-volatile because we check whether an attribute is non-volatile by
```
bool IsNonVolatile() const { return (mask & ATTRIBUTE_MASK_NONVOLATILE) && !IsExternal(); }
``` 
Can we have non-volatile attributes on dynamic endpoints?

 #### Proposed Solution
<suggested fix, suggested enhancement>

bzbarsky-apple, 2022-06-29:
When configuring attributes in ZAP, there are three mutually exclusive storage modes: RAM, non-volatile, external.  We  use two bits to represent these three mutually exclusive states.  The check represents the use of the bits we have.

The problems with introducing a fourth (external but non-volatile) state, especially for dynamic endpoint are as follows:

1.  The system that handles storing (and reading from storage) non-volatile values cannot handle external attributes that are handled via AttributeAccessInterface.  This might be ameliorated by not marking those as non-volatile....
2. The storage key for non-volatile attributes contains the endpoint id.  For dynamic endpoints, the endpoint id might not mean the same entity as time passes.

Fundamentally, for an external attribute, storage is up to the implementation of the external attribute...  If that implementation wants to use the same keys as "normal" non-volatile attribute storage, it can.

bzbarsky-apple, 2022-06-29:
In particular, `IsNonVolatile` clearly documents that it's about whether the attribute is _automatically_ stored in non-volatile memory.  But external attributes can be thus stored too, and the cluster impl should only be erroring out if it knows for sure the attributes are not being stored persistently....

bzbarsky-apple, 2022-06-30:
https://github.com/project-chip/connectedhomeip/pull/20144 should at least fix the issue @chiragatal describes.  The actual non-volatility is still the responsibility of the implementation of the external attribute.

closed 2022-06-30 by andy31415