project-chip/connectedhomeip #20100: [TC-MF-1.17] [TC-MF-1.18] [TC-MF-1.21] [TC-MF-1.22] Door Lock Application Hard Fault Error Because of credentialIndex used in GetCredential function

shgutte, 2022-06-29:
Problem: Door Lock Application Hard Fault Error Because of "credentialIndex" used in the "GetCredential" function during Multi Fabric cluster while removing fabric Index / while resetting the device 

Actual Behavior : 
function LockManager::GetCredential()
```
uint16_t adjustedCredentialIndex = credentialIndex - 1;
const auto & credentialInStorage = mLockCredentials[adjustedCredentialIndex];
```

This code is trying to create the array using the credentialIndex but if the credentialIndex is 0 and we do -1 from it then it's taking the max value of uint16_t that's why we are getting the "HARD FAULT" Error while doing the Muti-fabric Remove FabricIndex command and the same command used while resetting the device.

The range checking of credentialIndex is done in the door-lock-server.
In the door-lock-server.CPP

```
    // Programming PIN is a special case -- it is unique and its index is assumed to be 0.
    if (DlCredentialType::kProgrammingPIN == credentialType)
    {
        startIndex = 0;
        maxNumberOfCredentials--;
    }
```
[HardFault Error Device.txt](https://github.com/project-chip/connectedhomeip/files/9007624/HardFault.Error.Device.txt)

Expected Behavior : 
                It "credentialIndex" should not return 0 

Test case affecting because of this :
TC-MF-1.17
TC-MF-1.18
TC-MF-1.21
TC-MF-1.22

App used - lock app
Platform - Chip-tool - RPI-4, 8GB RAM
DUT - EFR32 + WF200 , EFR32 +RS9116 
Network - Ble-wifi

Logs:
Device side while resetting devices
[WF200-MF.txt](https://github.com/project-chip/connectedhomeip/files/9007829/WF200-MF.txt)

woody-apple, 2022-07-05:
Cert Blocker Review: Looks like this is a general issue in our example app.

woody-apple, 2022-07-05:
Cert Blocker Review: @Morozov-5F  are you able to take a look at this? Thanks!

closed 2022-07-12 by shgutte