project-chip/connectedhomeip #20118: [Pre-SVE] DUT is sending a QueryImage command before the minimum interval defined by spec

mamehta, 2022-06-29:
Cluster - software update (OTA)
Test case: TC-SU-2.5
Command number: steps 2 and 3
Expected: QueryImage must be sent only after the interval defined by spec (set by the delayedActionTime setting)
Actual Outcome: QueryImage is sent before the interval
DUT: EFR32 + RS9116 (WiFi)
<TODO: Attach logs>

bzbarsky-apple, 2022-06-30:
@carol-apple @selissia @tcarmelveilleux

tcarmelveilleux, 2022-07-05:
Can you please provide what the delay was that was provided by the provider and that was not honoured?

We need more logs

woody-apple, 2022-07-05:
Cert Blocker Review: Assigning to @cjandhyala to follow up.

chirag-silabs, 2022-07-07:
Sorry for the delay, here are the logs. The -p 180 but the transfer is happening without any delay

[chip-tool.txt](https://github.com/project-chip/connectedhomeip/files/9066156/chip-tool.txt)
[ota_dut_1.txt](https://github.com/project-chip/connectedhomeip/files/9066158/ota_dut_1.txt)
[ota_provider.txt](https://github.com/project-chip/connectedhomeip/files/9066159/ota_provider.txt)

andy31415, 2022-07-12:
@chirag-silabs - please provide more details, during request-cert-blocker review this seems to still needs to be disambiguated.

We need to understand what test steps are attempted, what spec section is verified and what behaviour difference exists between expected and actual.

selissia, 2022-07-12:
@chirag-silabs @mamehta:
TC-SU-2.5 Steps 2 and 3 refer to the  ApplyUpdateResponse command, not QueryImage. I am referring to the test document  Draft_TE9_TestPlanVerificationSteps_2020220609.xlsx . The -p parameter to chip-ota-provider-app app refers to ApplyUpdateResponse. 

The wording of the test case "Verify that the DUT starts updating its software" is a bit misleading. Here "updating its software" refers to applying the downloaded image, i.e. rebooting with this new image. It does not mean "start the whole software update process by issuing QueryImage"

mamehta, 2022-07-13:
I think this issue can be closed. I thought I saw a description expecting the DUT to send a QueryImage after receiving an ApplyUpdateResponse, but the latest test plan doesn't show that.

cjandhyala, 2022-07-13:
closing per @mamehta comments

closed 2022-07-13 by cjandhyala