project-chip/connectedhomeip #20143: Bootstrap is broken (looking for lib2to3 in the wrong places)

bzbarsky-apple, 2022-06-30:
#### Problem
After https://github.com/project-chip/connectedhomeip/pull/19992 bootstrap is broken: it looks for lib2to3, but even installing it with `pip3 install 2to3` does not help, because it looks for it in a (hardcoded) wrong place.

For now people are limping along by manually creating the wrong-place directory, but that's really not sane.

 #### Proposed Solution
We need to either roll back #19992 or get this fixed ASAP on the pigweed side.

@woody-apple @andy31415 @mspang @rgoliver

bzbarsky-apple, 2022-06-30:
Note that this is currently blocking me from doing some things, by the way, because I don't really want to mess with `sudo` and my system Python install to placate this broken script...

closed 2022-06-30 by andy31415

super-T-23, 2022-07-01:
Hi i met the same problem and pull latest pigweed version does not help either

PR #20158 (2022-06-30): Pull latest pigweed version.
```
cd third_party/pigweed/repo
git fetch https://github.com/google/pigweed.git main
git checkout FETCH_HEAD
```

Fixes #20143