project-chip/connectedhomeip #20149: TC-ACL-x-x Access control Verification Steps requires valid and invalid TLV data

aswathygrl, 2022-06-30:
[TC-ACL-2.3] Extension attribute
Test  Plan link: https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/AccessControl.adoc#323-tc-acl-23-extension-attribute

2. D_OK_EMPTY is an octstr of length 2 containing valid TLV: - top-level anonymous list (empty).

3. D_OK_SINGLE is an octstr of length 50-100 containing valid TLV: - top-level anonymous list, containing - one element with profile-specific tag in fully qualified form.

4. D_OK_FULL is an octstr of length 128 containing valid TLV: - top-level anonymous list, containing - two elements with profile-specific tag in fully qualified form.

5. D_BAD_LENGTH is an octstr of length 129 containing valid TLV: - top-level anonymous list, containing - one element with profile-specific tag in fully qualified form.

6. D_BAD_STRUCT is an octstr of length 2 containing valid TLV: - top-level anonymous struct, empty.

7. D_BAD_LIST is an octstr of length 50-100 containing valid TLV: - top-level list with context-specific tag, containing - one element with profile-specific tag in fully qualified form.

8. D_BAD_ELEM is an octstr of length 50-100 containing valid TLV: - top-level anonymous list, containing - one element with anonymous tag.

9. D_BAD_OVERFLOW is an octstr of length 50-100 containing valid TLV: - top-level anonymous list, containing - one element with profile-specific tag in fully qualified form - but having extra bytes after the top-level list end-of-container.

10. D_BAD_UNDERFLOW is an octstr of length 50-100 containing invalid TLV: - top-level anonymous list, containing - one element with profile-specific tag in fully qualified form - but lacking the top-level list end-of-container.

11. D_BAD_NONE is an octstr of length 0.

bzbarsky-apple, 2022-06-30:
@aswathygrl What exactly is the SDK issue here?

aswathygrl, 2022-07-01:
@bzbarsky-apple  This is not an SDK Issue ,we need the support from SDK to generate the TLV data listed in the above issue.

bzbarsky-apple, 2022-07-01:
@aswathygrl What exact support are you looking for?  Do you need to generate a single static set of payloads, or do you need to be generating these on the fly based on some input parameters?  If the latter, what input parameters?  What form do you need the data in (raw bytes, hex, base64, something else)?

vivien-apple, 2022-07-01:
My understanding of the ACL extension is that an ACL extension contains **arbitrary** TLV encoded data with the following constraints:
```The contents SHALL consist of a top-level anonymous list; each list element SHALL include a [profile-specific] tag encoded in fully-qualified form.```

So I suspect that there is no need to generates content on the fly, just static payloads.

cjandhyala, 2022-07-01:
@vivien-apple do we have tools to generate these payloads or do we need to build one ? Also does current chip-tool sending these TLV payloads ? Wondering if this test should be covered in unit tests or should this be a cert test. Since there is no SHALL requirement in the spec for these scenarios, I think these scenarios should be covered in unit tests rather than certification tests

bzbarsky-apple, 2022-07-01:
> Also does current chip-tool sending these TLV payloads ?

If you mean "support sending", then yes., you can set ACL extensions via chip-tool.

tcarmelveilleux, 2022-07-18:
We have means to generate this TLV. Both in Python and C++. Someone just has to do it once.

*This is not a unit-test case* --> The processing of extensions is per product and the core SDK does not itself process it, so we need to test the assumptions in cert

aswathygrl, 2022-07-25:
Closing this issue as the it is resoved and Updated the Test plan.

closed 2022-07-25 by aswathygrl