project-chip/connectedhomeip #20152: Running source scripts/activate.sh reports an error

nekteckIOS, 2022-06-30:
yuanrui_ios@YR-iOSdeMacBook-Pro connectedhomeip % source scripts/activate.sh

  WELCOME TO...

           ░▓░
           ▓█▓
           ▓█▓                                                     ▒█     ▒█
      ▒██▒▒▓██▒███▒                ░▒▓▒░  ░▒▓▒░       ░░▓█▒░ ░█  █████████████░    ░▒█▒░       ░░▒░
       ░▓█████▓██░               ▒█▒░░▒▓██▓▒░░▒█▒   ░█▓▒░░▒████    █▓░    █▓░    ▒█▒░░░▒█▓░   ██▒░░
    ▒█▒░         ░██▒           ░█░     █▓     ░█░ ░▓▒      ░▓█    █▓     █▓    ▒█░░    ░██  ▒▓
    ░▓██▓░     ░██▓█░           ░█      ▓█      █░ ░█░       ██    █▓     █▓    ▓██████████  ▒█
     ░▓███▒   ▒███▒░            ░█      ▓█      █░  ██░    ░███    █▓     █▓    ░█▒░         ▒█
 ░▒████████░ ░███▓▓█▓▓▒         ░█      ▒▒      █░   ░█▓██▓█░▒█    ░▓▓█░  ░▓▓█░   ▒▓▓██▓█░   ▒▓
 ░██▒░  ▒██▒ ▒██░  ░▒█▓
         ░▓░ ░▓░

  BOOTSTRAP! Bootstrap may take a few minutes; please be patient.

Downloading and installing packages into local source directory:

  Setting up CIPD package manager...[-][P15367 16:39:33.496 resolver.go:175 D] Resolving package gn/gn/mac-arm64@git_revision:29accf5ac2eadfc53e687081583b7bc1592a8839 ...
[P15367 16:39:33.496 resolver.go:175 D] Resolving package pigweed/host_tools/mac-arm64@git_revision:876b235af6456367c18efb5b6f6425fbbe9de404 ...
[P15367 16:39:33.497 resolver.go:175 D] Resolving package fuchsia/third_party/qemu/mac-arm64@version:v6.2.0 ...
[P15367 16:39:33.497 resolver.go:175 D] Resolving package infra/3pp/tools/buildifier/mac-arm64@version:2@5.1.0 ...
[P15367 16:39:33.497 resolver.go:175 D] Resolving package infra/3pp/tools/ninja/mac-arm64@version:2@1.11.0.chromium.4 ...
[P15367 16:39:33.497 client.go:918 D] Tag cache hit for gn/gn/mac-arm64:git_revision:29accf5ac2eadfc53e687081583b7bc1592a8839 - 4_0tZ1u_on6zcJIJazvLhlv791AATWp-sywLn76t5nEC
[P15367 16:39:33.497 resolver.go:178 D] Resolved package gn/gn/mac-arm64@git_revision:29accf5ac2eadfc53e687081583b7bc1592a8839 => gn/gn/mac-arm64:4_0tZ1u_on6zcJIJazvLhlv791AATWp-sywLn76t5nEC
[P15367 16:39:33.497 client.go:918 D] Tag cache hit for infra/3pp/tools/ninja/mac-arm64:version:2@1.11.0.chromium.4 - EIes683ruLLnes38SfkIvrrbIqZhflwR5F1C7PBa3lAC
[P15367 16:39:33.497 resolver.go:178 D] Resolved package infra/3pp/tools/ninja/mac-arm64@version:2@1.11.0.chromium.4 => infra/3pp/tools/ninja/mac-arm64:EIes683ruLLnes38SfkIvrrbIqZhflwR5F1C7PBa3lAC
[P15367 16:39:33.497 client.go:918 D] Tag cache hit for infra/3pp/tools/buildifier/mac-arm64:version:2@5.1.0 - ZJiaja8OTZLfIEnx_Rb1NUYfBV-acyk8p9RdYFtQXTsC
[P15367 16:39:33.497 resolver.go:178 D] Resolved package infra/3pp/tools/buildifier/mac-arm64@version:2@5.1.0 => infra/3pp/tools/buildifier/mac-arm64:ZJiaja8OTZLfIEnx_Rb1NUYfBV-acyk8p9RdYFtQXTsC
[P15367 16:39:33.497 resolver.go:175 D] Resolving package infra/3pp/tools/go/mac-arm64@version:2@1.18.3 ...
[P15367 16:39:33.497 client.go:424 D] RPC chrome-infra-packages.appspot.com/cipd.Repository.ResolveVersion {"host":"chrome-infra-packages.appspot.com", "method":"ResolveVersion", "service":"cipd.Repository"}
[P15367 16:39:33.497 client.go:918 D] Tag cache hit for infra/3pp/tools/go/mac-arm64:version:2@1.18.3 - YxNK10MWb2qDS3watve8XNUqE8uk7g4Pp7GvFVbbprwC
[P15367 16:39:33.497 resolver.go:178 D] Resolved package infra/3pp/tools/go/mac-arm64@version:2@1.18.3 => infra/3pp/tools/go/mac-arm64:YxNK10MWb2qDS3watve8XNUqE8uk7g4Pp7GvFVbbprwC
[P15367 16:39:33.497 resolver.go:175 D] Resolving package pigweed/third_party/gcc-arm-none-eabi/mac-arm64@version:10-2020-q4-major ...
[P15367 16:39:33.497 resolver.go:175 D] Resolving package fuchsia/third_party/cmake/mac-arm64@version:3.24.20220625-g42984e7 ...
[P15367 16:39:33.497 resolver.go:175 D] Resolving package infra/3pp/tools/openocd/mac-arm64@version:2@0.11.0-3 ...
[P15367 16:39:33.497 client.go:918 D] Tag cache hit for infra/3pp/tools/openocd/mac-arm64:version:2@0.11.0-3 - sU-sLAqal-ve3NiwxR_fZn5pgspqsGRS04BhZ5vodBoC
[P15367 16:39:33.497 resolver.go:178 D] Resolved package infra/3pp/tools/openocd/mac-arm64@version:2@0.11.0-3 => infra/3pp/tools/openocd/mac-arm64:sU-sLAqal-ve3NiwxR_fZn5pgspqsGRS04BhZ5vodBoC
[P15367 16:39:33.497 resolver.go:175 D] Resolving package pigweed/third_party/protoc-gen-go/mac-arm64@version:1.3.2 ...
[P15367 16:39:33.497 client.go:424 D] RPC chrome-infra-packages.appspot.com/cipd.Repository.ResolveVersion {"host":"chrome-infra-packages.appspot.com", "method":"ResolveVersion", "service":"cipd.Repository"}
[P15367 16:39:33.497 resolver.go:175 D] Resolving package pigweed/third_party/bloaty-embedded/mac-arm64@git_revision:2d87d204057b419f5290f8d38b61b9c2c5b4fb52-2 ...
[P15367 16:39:33.497 resolver.go:175 D] Resolving package infra/rbe/client/mac-arm64@re_client_version:0.58.0.d457d5a ...
[P15367 16:39:33.497 resolver.go:175 D] Resolving package infra/3pp/tools/cpython3/mac-arm64@version:2@3.9.5.chromium.19 ...
[P15367 16:39:33.497 resolver.go:175 D] Resolving package fuchsia/third_party/cmake/mac-arm64@version:3.20.20210428-g857373c ...
[P15367 16:39:33.497 resolver.go:175 D] Resolving package infra/3pp/tools/protoc/mac-arm64@version:2@3.17.3 ...
[P15367 16:39:33.497 client.go:424 D] RPC chrome-infra-packages.appspot.com/cipd.Repository.ResolveVersion {"host":"chrome-infra-packages.appspot.com", "method":"ResolveVersion", "service":"cipd.Repository"}
[P15367 16:39:33.497 resolver.go:175 D] Resolving package fuchsia/third_party/clang/mac-arm64@git_revision:1aa59ff2f789776ebfa2d4b315fd3ea589652b4a ...
[P15367 16:39:33.497 client.go:424 D] RPC chrome-infra-packages.appspot.com/cipd.Repository.ResolveVersion {"host":"chrome-infra-packages.appspot.com", "method":"ResolveVersion", "service":"cipd.Repository"}
[P15367 16:39:33.497 client.go:424 D] RPC chrome-infra-packages.appspot.com/cipd.Repository.ResolveVersion {"host":"chrome-infra-packages.appspot.com", "method":"ResolveVersion", "service":"cipd.Repository"}
[P15367 16:39:33.498 client.go:424 D] RPC chrome-infra-packages.appspot.com/cipd.Repository.ResolveVersion {"host":"chrome-infra-packages.appspot.com", "method":"ResolveVersion", "service":"cipd.Repository"}
[P15367 16:39:33.498 client.go:918 D] Tag cache hit for infra/rbe/client/mac-arm64:re_client_version:0.58.0.d457d5a - K8ZLY7IaheEhLgljRFeSOHa4J0rTtTmh7_r1_yNUd4YC
[P15367 16:39:33.498 resolver.go:178 D] Resolved package infra/rbe/client/mac-arm64@re_client_version:0.58.0.d457d5a => infra/rbe/client/mac-arm64:K8ZLY7IaheEhLgljRFeSOHa4J0rTtTmh7_r1_yNUd4YC
[P15367 16:39:33.498 client.go:424 D] RPC chrome-infra-packages.appspot.com/cipd.Repository.ResolveVersion {"host":"chrome-infra-packages.appspot.com", "method":"ResolveVersion", "service":"cipd.Repository"}
[P15367 16:39:33.498 client.go:424 D] RPC chrome-infra-packages.appspot.com/cipd.Repository.ResolveVersion {"host":"chrome-infra-packages.appspot.com", "method":"ResolveVersion", "service":"cipd.Repository"}
[P15367 16:39:33.498 client.go:424 D] RPC chrome-infra-packages.appspot.com/cipd.Repository.ResolveVersion {"host":"chrome-infra-packages.appspot.com", "method":"ResolveVersion", "service":"cipd.Repository"}
[P15367 16:39:33.498 client.go:424 D] RPC chrome-infra-packages.appspot.com/cipd.Repository.ResolveVersion {"host":"chrome-infra-packages.appspot.com", "method":"ResolveVersion", "service":"cipd.Repository"}
[P15367 16:39:33.920 resolver.go:180 D] Failed to resolve package fuchsia/third_party/clang/mac-arm64@git_revision:1aa59ff2f789776ebfa2d4b315fd3ea589652b4a: no such package: fuchsia/third_party/clang/mac-arm64
[P15367 16:39:33.925 resolver.go:180 D] Failed to resolve package pigweed/host_tools/mac-arm64@git_revision:876b235af6456367c18efb5b6f6425fbbe9de404: no such package: pigweed/host_tools/mac-arm64
[P15367 16:39:33.928 resolver.go:180 D] Failed to resolve package fuchsia/third_party/cmake/mac-arm64@version:3.20.20210428-g857373c: no such package: fuchsia/third_party/cmake/mac-arm64
[P15367 16:39:33.928 resolver.go:180 D] Failed to resolve package fuchsia/third_party/cmake/mac-arm64@version:3.24.20220625-g42984e7: no such package: fuchsia/third_party/cmake/mac-arm64
[P15367 16:39:33.931 resolver.go:180 D] Failed to resolve package pigweed/third_party/gcc-arm-none-eabi/mac-arm64@version:10-2020-q4-major: no such package: pigweed/third_party/gcc-arm-none-eabi/mac-arm64
[P15367 16:39:33.931 resolver.go:180 D] Failed to resolve package infra/3pp/tools/cpython3/mac-arm64@version:2@3.9.5.chromium.19: no such tag
[P15367 16:39:33.931 resolver.go:180 D] Failed to resolve package infra/3pp/tools/protoc/mac-arm64@version:2@3.17.3: no such tag
[... cut ...]
subprocess.CalledProcessError: Command '['/Users/yuanrui_ios/connectedhomeip/.environment/cipd/cipd', 'ensure', '-ensure-file', '/Users/yuanrui_ios/connectedhomeip/.environment/cipd/packages.ensure', '-root', '/Users/yuanrui_ios/connectedhomeip/.environment/cipd/packages', '-log-level', 'debug', '-json-output', '/Users/yuanrui_ios/connectedhomeip/.environment/cipd/packages.json', '-cache-dir', '/Users/yuanrui_ios/.cipd-cache-dir', '-max-threads', '0']' returned non-zero exit status 1.

Dear Team,
When I execute source **scripts/activate.sh**, an error will be reported. What should I do to make it run normally?

bzbarsky-apple, 2022-06-30:
@andy31415 @mspang @rgoliver

wickyou23, 2022-07-01:
I got the same error yesterday.

wickyou23, 2022-07-01:
Today, when I pull the latest commit, it works well.

nekteckIOS, 2022-07-01:
> I got the same error yesterday. Today, when I pull the latest commit, it works well.

Are you running in the M1 environment?

wickyou23, 2022-07-01:
> > I got the same error yesterday. Today, when I pull the latest commit, it works well.
> 
> Are you running in the M1 environment?

Yes, I am running in the M1 chip.

closed 2022-07-01 by nekteckIOS

TheWalkingDead1024, 2022-09-02:
got the same issue