project-chip/connectedhomeip #20160: Seeing ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0 on Nordic logs when left running for few hours

vinay-apple, 2022-06-30:
#### Problem

Seeing ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0 on Nordic logs when left running for few hours

SHA: 5d8599d195a93dfe5473f64e4f888c322e5df1d2

 * steps to reproduce
1. Commission Nordic thread device in Apple Home app and let it sit for few hour (7-8 hr) and seeing that it goes unreachable in Apple Home app and seeing below fatal error on Nordic logs. Detailed logs added. 

  
ASSERTION FAIL [!arch_is_in_isr()] @ WEST_TOPDIR/zephyr/kernel/mutex.c:101
        mutexes cannot be used inside ISRs
E: r0/a1:  0x00000004  r1/a2:  0x00000065  r2/a3:  0x200075c8
E: r3/a4:  0x00000000 r12/ip:  0x00000000 r14/lr:  0x00048797
E:  xpsr:  0x61000021
E: s[ 0]:  0x00000000  s[ 1]:  0x00000000  s[ 2]:  0x00000000  s[ 3]:  0x00000000
E: s[ 4]:  0x00000000  s[ 5]:  0x00000000  s[ 6]:  0x00000000  s[ 7]:  0x00000000
E: s[ 8]:  0x00000000  s[ 9]:  0x00000000  s[10]:  0x00003400  s[11]:  0x00000000
E: s[12]:  0x00003408  s[13]:  0x00000000  s[14]:  0x00000000  s[15]:  0x00000000
E: fpscr:  0x200075c8
E: Faulting instruction address (r15/pc): 0x0009dcbe
E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
E: Fault during interrupt handling

E: Current thread: 0x200075c8 (idle)
E: Halting system

[Nordic-Fatal-Error-logs.txt](https://github.com/project-chip/connectedhomeip/files/9020384/Nordic-Fatal-Error-logs.txt)

bzbarsky-apple, 2022-06-30:
@Damian-Nordic @kkasperczyk-no

Damian-Nordic, 2022-06-30:
@vinay-apple Which example is that?

Damian-Nordic, 2022-06-30:
I have a suspicion that it is caused by the timer used to store TotalOperatingHours in non-volatile storage. The operation is triggered by Zephyr's low-level timer (the handler is called from ISR) every 10 hours by default and puts a task onto the CHIP thread which involves `write` syscall that locks a mutex. It used to work a bit differently, but after the recent Zephyr update may lead to the crash. We'll fix that tomorrow. Thanks for reporting!

vinay-apple, 2022-06-30:
@Damian-Nordic was using the lighting app

PR #20214 (2022-07-01): [zephyr] Fix crash on saving TotalOperatingHours to KVS
#### Problem
Calling PlatformMgr().PostEvent() from ISR, such as a Zephyr timer handler, would no longer work after a recent Zephyr upmerge that added taking a mutex to write() syscalls.

#### Change overview
Rework the method to work even when called from the timer handler.
Fixes #19579
Fixes #20160

#### Testing
Verified with an edited example which stores TotalOperatingHours every minute, that nRF Connect examples crash without and they don't with the change.
closed 2022-07-01 by andy31415