project-chip/connectedhomeip #20210: Ensure ESP32 storage backend succeeds Storage audit

shubhamdp, 2022-07-01:
#### Problem
PR #20134 and #20164 together introduced a storage API compliance audit.

This audit must be run and pass on ESP32 and succeed against PersistentStorageDelegate used by all-clusters-app.

To run the audit for anything using Server, build your target with `chip_support_enable_storage_api_audit=true` added to your GN args and then run example. The default audit is added to Server::Init() and will cause an assert with logs on failure.

Audit passes once you see the following string in your logs

==== PersistentStorageDelegate API audit: SUCCESS ====
On failure, informative assertion failures are logged to assist in debugging.

 #### Proposed Solution
* Build with audit enabled
* Run and comment on passing along with the SHA of the commit where it was run. Provide a logs capture of your run.
* If you find failures, please document them in comments on the issue to help SDK team audit the types of failures/assumptions made.
* Close the issue once audit passes

tcarmelveilleux, 2022-07-02:
> #20211 Fixes 4 out of 5 and now theres only one error
> 
> ```
> E (1686) chip[ATM]: ../../../../../../config/esp32/third_party/connectedhomeip/src/lib/support/PersistentStorageAudit.cpp:145: assertion failed: "size == sizeBeforeGetKeyValueCall"
> 
> E (1726) chip[ATM]: ==== PersistentStorageDelegate API audit: FAILED: 1/81 failed assertions ====
> ```
> 
> This is because ESP-IDF NVS APIs do not support partial reads. :-(

> #20211 Fixes 4 out of 5 and now theres only one error
> 
> ```
> E (1686) chip[ATM]: ../../../../../../config/esp32/third_party/connectedhomeip/src/lib/support/PersistentStorageAudit.cpp:145: assertion failed: "size == sizeBeforeGetKeyValueCall"
> 
> E (1726) chip[ATM]: ==== PersistentStorageDelegate API audit: FAILED: 1/81 failed assertions ====
> ```
> 
> This is because ESP-IDF NVS APIs do not support partial reads. :-(

- Could ESP-IDF NVS API made to support partial reads (the answer is usually yes)
- Can ESP-IDF NVS API allow determining the size of a value without a full read of it?

shubhamdp, 2022-07-02:
- I will get it done, just that it would be available in next release.
- Yes, we do get the size of value without full read. I had this though: 
    - Read the size and allocate that number of bytes
    - Copy the requested bytes to the out_buffer and return `CHIP_ERROR_TOO_SMALL`

But I was taken aback with the though of doing dynamic allocation. If malloc seems okay, then for now I can implement the second option and open up an issue for using partial read API when it will be available.

tcarmelveilleux, 2022-07-02:
We should not have to read the full key and truncate it. The API should use the storage provided by the caller.

I noticed as well in `src/platform/ESP32/KeyValueStoreManagerImpl.cpp` that you are using hashing when the key is long. This is problematic since it's not impossible (although unlikely) to have key collisions, and you would never know. If you use hashing, you have to be able to determine if at the hashed key, you are storing/loading the equivalent of the original key...

tehampson, 2022-07-04:
>This is because ESP-IDF NVS APIs do not support partial reads

Mind linking to the APIs? Is it that if you give a size that is too small it won't do read up to that size?

closed 2023-01-20 by stale[bot]