project-chip/connectedhomeip #20215: [Android] 'Unresolved reference: chip' error on Android Studio - Mac Mini M1.

wickyou23, 2022-07-01:
#### Problem

Hello everyone, 

Currently, I followed the document and successfully built and ran the android app in debug mode, but I got "Unresolved reference: chip" error in Android Studio. I don't know if I missed any steps?

Thanks.

See the image for details:

[image]

More information:
- PC: Mac mini M1.
- NDK version: 21.4.7075529.
- CMake version: 3.10.2.4988404.
- JDK version: 11.0.12

matdev, 2022-09-19:
Same issue. Any update on this ?

fangxiaoying, 2022-09-29:
same issue.

rrosasm, 2022-09-30:
Running into this as well. Has anyone worked past it?

rrosasm, 2022-10-11:
I am no longer seeing this particular issue on the v1.0-branch

GnarFang, 2023-06-14:
same issue