project-chip/connectedhomeip #20218: [TC-MF-1.17/18/19/20] - Multi-admin scenario fails

sowmyassp, 2022-07-01:
**Issue :** When I commission DUT with 3 controllers with same fabric and later remove the second controllers fabric and if I pass avahi-browse -rt _matter._tcp I am getting 3 Operational service records instead 2.

**Commit used :** c38e9159d0500a880b9c789bf5d87fa16c039633 			

**Expected behavior :** There should be 2 Operational service records when we pass "avahi-browse -rt _matter._tcp" command

**Actual behavior :** There is 3 Operational service records

**Steps to reproduce :**

- DUT side :
sudo rm -rf /tmp/chip_*
sudo ./chip-all-clusters-app --wifi

- TH side :
1. ./chip-tool pairing ble-wifi 1 zigbeehome matter123 20202021 3840 (on TH1)
2. ./chip-tool pairing open-commissioning-window 1 1 900 1000 3840(on TH1)
3. ./chip-tool pairing manualcode 2 36366524220(on TH2)
4. ./chip-tool pairing open-commissioning-window 1 1 900 1000 3840(on TH1)
5. ./chip-tool pairing manualcode 2 36237314697(on TH3)
6. ./chip-tool operationalcredentials read fabrics 2 0 --fabric-filtered 0(on TH2)
7. ./chip-tool operationalcredentials remove-fabric 2 2 0(on TH2)
8. avahi-browse -rt _matter._tcp

**Test platform :**
Chip-tool - RPI-4, 8GB RAM
DUT - RPI - RPI-4, 8GB RAM

**Test-plan reference:**
https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/multiplefabrics.adoc#tc-mf-1-17-removing-fabrics-from-dut-and-fabric-index-enumeration-using-ecm-dut-commissionee

bzbarsky-apple, 2022-07-02:
@andy31415 I thought we were resetting things with a zero TTL when using minimal nowadays?

But past that @sowmyassp it's entirely possible that there are stale DNS entries cached by someone, until the TTL expires.  Doing a browse like that is not a reliable way to tell what the device is actually advertising.

sowmyassp, 2022-07-05:
@bzbarsky-apple Test-plan says so
 "Verify DUT_CE is now discoverable over DNS-SD with 2 Operational service records (_matter._tcp SRV records)"

bzbarsky-apple, 2022-07-12:
@sowmyassp The test plan can say whatever it wants, but that's not going to change how DNS and TTLs work...  Stale records can happen, until the TTL expires.

andy31415, 2022-07-12:
We are setting zero TTL as far as I am aware.
That being said, is the test correct? It assumes no stale DNSSD records.

andy31415, 2022-07-12:
@cjandhyala @franck-apple to tag this appropriately: testing for this is valid, however proposed steps do not work. Need differerent approach as described above.

cjandhyala, 2022-07-13:
@sowmyassp , why are you using same node ID in Step 5 and step 3