project-chip/connectedhomeip #20242: Continuously send commands to the device, the device cannot allocate space

ChengqiangShe, 2022-07-02:
commit： 6efefe37c64bf92bde0a52de7994f377e2d9bc2d

 #### Problem
Continuously send commands to the device, the device cannot allocate space

application:
connectedhomeip/examples/lighting-app/efr32

log：
```
00> <detail> [DMG] IM WH moving to [AddStatus]
00> <detail> [EM] Piggybacking Ack for MessageCounter:8959649 on exchange: 43145r
00> <detail> [IN] UDP Message Received packet nb : 9943 SrcAddr : 2001:db8:0:1:77d3:a83b:b523:22dc[5540] DestAddr : fd4c:1eb9:a280:1:f767:5e09:3766:21e1[5540] Payload Length 34
00> <info  > [IN] Prepared secure message 0x2001e69c to 0x000000000012D693 (1)  of type 0x7 and protocolId (0, 1) on exchange 43145r with MessageCounter:61293062.
00> <info  > [IN] Sending encrypted msg 0x2001e69c with MessageCounter:61293062 to 0x000000000012D693 (1) at monotonic time: 00000000000EBB79 msec
00> <detail> [IN] UDP Message Received packet nb : 9944 SrcAddr : 2001:db8:0:1:77d3:a83b:b523:22dc[5540] DestAddr : fd4c:1eb9:a280:1:f767:5e09:3766:21e1[5540] Payload Length 34
00> <detail> [DMG] IM WH moving to [Sending]
00> <detail> [DMG] IM WH moving to [Uninitialized]
00> <error > [DL] Long dispatch time: 102 ms, for event type 3
00> <info  > [EM] Received message of type 0x6 with protocolId (0, 1) and MessageCounter:8959650 on exchange 43146r
00> <detail> [EM] Handling via exchange: 43146r, Delegate: 0x2000a6ec
00> <detail> [IM] Received Write request
00> <detail> [DMG] IM WH moving to [Initialized]
00> <error > [CSL] PacketBuffer: pool EMPTY.
00> <detail> [DMG] IM WH moving to [Uninitialized]
00> <error > [CSL] PacketBuffer: pool EMPTY.
00> <error > [CSL] PacketBuffer: pool EMPTY.
00> <error > [CSL] PacketBuffer: pool EMPTY.
00> <error > [IN] Failed to allocate a System buffer of size 62 for UDP Message reception.
00> <error > [CSL] PacketBuffer: pool EMPTY.
00> <error > [IN] Failed to allocate a System buffer of size 34 for UDP Message reception.
00> <error > [CSL] PacketBuffer: pool EMPTY.
00> <error > [IN] Failed to allocate a System buffer of size 62 for UDP Message reception.
00> <error > [CSL] PacketBuffer: pool EMPTY.
00> <error > [IN] Failed to allocate a System buffer of size 62 for UDP Message reception.
00> <error > [CSL] PacketBuffer: pool EMPTY.
00> <error > [IN] Failed to allocate a System buffer of size 62 for UDP Message reception.
00> <error > [CSL] PacketBuffer: pool EMPTY.
00> <error > [IN] Failed to allocate a System buffer of size 62 for UDP Message reception.
00> <error > [CSL] PacketBuffer: pool EMPTY.
00> <error > [IN] Failed to allocate a System buffer of size 62 for UDP Message reception.
00> <error > [CSL] PacketBuffer: pool EMPTY.
00> <error > [IN] Failed to allocate a System buffer of size 62 for UDP Message reception.
00> <error > [CSL] PacketBuffer: pool EMPTY.
00> <error > [IN] Failed to allocate a System buffer of size 62 for UDP Message reception.
00> <error > [CSL] PacketBuffer: pool EMPTY.
00> <error > [IN] Failed to allocate a System buffer of size 62 for UDP Message reception.
00> <error > [CSL] PacketBuffer: pool EMPTY.
00> <error > [IN] Failed to allocate a System buffer of size 62 for UDP Message reception.
00> <error > [CSL] PacketBuffer: pool EMPTY.
00> <error > [IN] Failed to allocate a System buffer of size 62 for UDP Message reception.
00> <error > [CSL] PacketBuffer: pool EMPTY.
00> <error > [IN] Failed to allocate a System buffer of size 62 for UDP Message reception.
00> <error > [CSL] PacketBuffer: pool EMPTY.
00> <error > [IN] Failed to allocate a System buffer of size 62 for UDP Message reception.
00> <error > [CSL] PacketBuffer: pool EMPTY.
00> <error > [IN] Failed to allocate a System buffer of size 62 for UDP Message reception.
00> <error > [CSL] PacketBuffer: pool EMPTY.
00> <error > [IN] Failed to allocate a System buffer of size 62 for UDP Message reception.
00> <error > [CSL] PacketBuffer: pool EMPTY.
00> <error > [IN] Failed to allocate a System buffer of size 62 for UDP Message reception.
00> <error > [CSL] PacketBuffer: pool EMPTY.
00> <error > [IN] Failed to allocate a System buffer of size 62 for UDP Message reception.
00> <error > [CSL] PacketBuffer: pool EMPTY.
00> <error > [IN] Failed to allocate a System buffer of size 62 for UDP Message reception.
00> <error > [CSL] PacketBuffer: pool EMPTY.
00> <error > [IN] Failed to allocate a System buffer of size 62 for UDP Message reception.
00> <error > [CSL] PacketBuffer: pool EMPTY.
00> <error > [IN] Failed to allocate a System buffer of size 62 for UDP Message reception.
00> <error > [CSL] PacketBuffer: pool EMPTY.
[... 182 lines cut ...]
00> <error > [IN] Failed to allocate a System buffer of size 62 for UDP Message reception.
00> <error > [CSL] PacketBuffer: pool EMPTY.
00> <error > [IN] Failed to allocate a System buffer of size 62 for UDP Message reception.
00> <error > [CSL] PacketBuffer: pool EMPTY.
00> <error > [IN] Failed to allocate a System buffer of size 62 for UDP Message reception.
00> <error > [CSL] PacketBuffer: pool EMPTY.
00> <error > [IN] Failed to allocate a System buffer of size 62 for UDP Message reception.
00> <error > [CSL] PacketBuffer: pool EMPTY.
00> <error > [IN] Failed to allocate a System buffer of size 62 for UDP Message reception.
00> <error > [CSL] PacketBuffer: pool EMPTY.
00> <error > [IN] Failed to allocate a System buffer of size 62 for UDP Message reception.
00> <error > [CSL] PacketBuffer: pool EMPTY.
00> <error > [IN] Failed to allocate a System buffer of size 62 for UDP Message reception.
00> <error > [CSL] PacketBuffer: pool EMPTY.
00> <error > [IN] Failed to allocate a System buffer of size 62 for UDP Message reception.
00> <error > [CSL] PacketBuffer: pool EMPTY.
00> <error > [IN] Failed to allocate a System buffer of size 62 for UDP Message reception.
00> <error > [CSL] PacketBuffer: pool EMPTY.
00> <error > [IN] Failed to allocate a System buffer of size 62 for UDP Message reception.
00> <error > [CSL] PacketBuffer: pool EMPTY.
00> <error > [IN] Failed to allocate a System buffer of size 62 for UDP Message reception.
00> <error > [CSL] PacketBuffer: pool EMPTY.
00> <error > [IN] Failed to allocate a System buffer of size 62 for UDP Message reception.
00> <error > [CSL] PacketBuffer: pool EMPTY.
```

ChengqiangShe, 2022-07-02:
[efr32_master_020220702_lighting.log](https://github.com/project-chip/connectedhomeip/files/9032496/efr32_master_020220702_lighting.log)

ChengqiangShe, 2022-07-02:
@jmartinez-silabs

jmartinez-silabs, 2022-07-04:
@ChengqiangShe Can you provide a way to reproduce this issue? I am not hitting this

ChengqiangShe, 2022-07-05:
You can use chip-tool to continuously read the onoff Attribute at intervals of 150ms. It will appear in about 10 minutes.

jepenven-silabs, 2022-07-05:
This is weird since our Packet Buffer pool is of size 8 as can be seen in `src/platform/EFR32/SystemPlatformConfig.h` I would have expected a buffer allocation failure a lot sooner

ChengqiangShe, 2022-08-03:
In the sve branch code test on July 30, this problem did not occur

closed 2022-08-03 by ChengqiangShe