project-chip/connectedhomeip #20279: Pre-SVE  [TC-OPCREDS-3.1] #4 and #6  cannot be verified in chip-tool verification steps as certificate content is not printed in chip-tool log of commissioner

morahman2022, 2022-07-04:
**Test Case ID**: TC-OPCREDS-3.1

**Describe Issue**:

TC-OPCREDS-3.1 verification steps for #4 and #6 of  in the test plan ( https://docs.google.com/spreadsheets/d/1h4d4J90KAxFqUL_cXZHyM3pHjHCEXzFITkQhN2KMqDY/edit#gid=66916293 ) mention that certificate will be printed in chip-tool log. 

Executing #4 and #6 involves commissioning an end device in #2 and checking certificate in the commissioner (chip-tool)  log. #4 looks for content of certificate while #6 verifies that the size of certificate is less than or equal to 600 bytes and of type octstr.

We commissioned an open-thread end device with BRD4187A and lighting app in #2 executing the command
`./chip-tool pairing ble-thread 1234 hex:0e080000000000000000000300000b35060004001fffe00208dead00beef00cafe0708fddead00beef0000051007955a7234f1d50fee29ebfab6b260b7030a4f70656e546872656164010242ee0410732879329cd7221ef7c19addc8be5fe10c0402a0fff8 20202021 3840`

The end device was commissioned successfully. However, chip-tool log did not print content of certificate for verification in #4 and #6. chip-tool log TC-OPCREDS-3.1_2.txt is attached with this bug submission.

[TC-OPCREDS-3.1_2.txt](https://github.com/project-chip/connectedhomeip/files/9040840/TC-OPCREDS-3.1_2.txt)

andy31415, 2022-07-12:
Cert blocker review: cannot easily evaluate this issue based on provided info. Please provide more information so we can figure out what is going on (cert/spec issue vs test step updates).

For more debugging and certificate validation, `--trace_decode 1` appended to chip-tool helps us in seeing certificate data in logs.

cjandhyala, 2022-07-13:
@morahman2022 , as marked in the SVE TC List, this TC was not ready to be executed and was marked as so. Pls execute this TC when its ready. Closing the issue for now.

closed 2022-07-13 by cjandhyala