project-chip/connectedhomeip #20282: [Pre-SVE] [TC-G-2.3] Read Group Table Returns 0 entries

mamehta, 2022-07-04:
#### Problem
[TC-G-2.3] Commands - GetGroupMembership, AddGroupIfIdentifying [DUT-Server]->TC-G-8b
Command run after adding two groups: "./chip-tool groupkeymanagement read group-table 1 0"

Expected Outcome:
we should get the output as 
[1653552886.221787][2625:2630] CHIP:TOO:   GroupTable: 2 entries
[1653552886.221925][2625:2630] CHIP:TOO:     [1]: {
[1653552886.221966][2625:2630] CHIP:TOO:       GroupId: 2
[1653552886.222002][2625:2630] CHIP:TOO:       Endpoints: 0 entries
[1653552886.222057][2625:2630] CHIP:TOO:       GroupName: gp2
[1653552886.222103][2625:2630] CHIP:TOO:       FabricIndex: 1
[1653552886.222135][2625:2630] CHIP:TOO:      }
[1653552886.222177][2625:2630] CHIP:TOO:     [2]: {
[1653552886.222222][2625:2630] CHIP:TOO:       GroupId: 3
[1653552886.222255][2625:2630] CHIP:TOO:       Endpoints: 0 entries
[1653552886.222288][2625:2630] CHIP:TOO:       GroupName: gp3
[1653552886.222330][2625:2630] CHIP:TOO:       FabricIndex: 1
[1653552886.222361][2625:2630] CHIP:TOO:      }

Actual Outcome:
We are getting the output as-
CHIP:TOO: Endpoint: 0 Cluster: 0x0000_003F Attribute 0x0000_0001 DataVersion: 1132406184
[1656531240.500102][3051:3056] CHIP:TOO:   GroupTable: 0 entries
[1656531240.500164][3051:3056] CHIP:EM: Sending Standalone Ack for MessageCounter:91298527 on exchange 46399i
[1656531240.500222][3051:3056] CHIP:IN: Prepared secure message 0x7fafffd828 to 0x0000000000000001 (1)  of type 0x10 and protocolId (0, 0) on exchange 46399i with MessageCounter:52118073.

We should get the Group entries as 2 but we are getting as zero 
<TODO: Attach logs>

rcasallas-silabs, 2022-07-05:
The observed behavior is correct. According to the spec, section 11.2.7.3. (GroupTable Attribute):
```
The GroupTable SHALL NOT contain any entry whose GroupInfoMapStruct has an empty Endpoints list. If a RemoveGroup or RemoveAllGroups command causes the removal of a group mapping from its last mapped endpoint, the entire GroupTable entry for that given GroupId SHALL be removed.
```

jepenven-silabs, 2022-07-05:
Re-opening since reported also by @sethunk . Currently investigating it since this function wasn't tested before and it's a combination of two cluster (Group and Identify)

jepenven-silabs, 2022-07-05:
Test plan Issue. Managed to get it to work on my side without any SDK modification

jepenven-silabs, 2022-07-05:
Closing since Test plan issue https://github.com/CHIP-Specifications/chip-test-plans/issues/1649

closed 2022-07-05 by jepenven-silabs