project-chip/connectedhomeip #20283: Pre-SVE [TC-BINFO-2.2] #2a and#4 cannot be exectued. #6 is optional and was not executed. the output of chip-tool command and RTT log for DUT do not match the expected outcome for #2b.

morahman2022, 2022-07-04:
**Test Case ID**: TC-BINFO-2.2

**Describe Issue**:

End devices with BRD-4187A and BRD-4161A are considered for testing TC-BINFO-2.2. lighting app was used.

#2a cannot be executed as graceful shutdown is not supported in the end devices.
#2b outputs from chip-tool command and J-Link RTT Viewer log for DUT do not match the expected outcome mentioned in test plan ( https://docs.google.com/spreadsheets/d/1h4d4J90KAxFqUL_cXZHyM3pHjHCEXzFITkQhN2KMqDY/edit#gid=66916293 ).
#4 outputs in the test plan could not be reproduced with factory resetting the end device.
#6 cannot be executed as there is no way to change the `Reachable` attribute. Moreover, #6 is optional.

[TC-BINFO-2.2_2b.txt](https://github.com/project-chip/connectedhomeip/files/9041407/TC-BINFO-2.2_2b.txt)
[TC-BINFO-2.2_2b_DUT_RTT_LOG.txt](https://github.com/project-chip/connectedhomeip/files/9041409/TC-BINFO-2.2_2b_DUT_RTT_LOG.txt)

bzbarsky-apple, 2022-07-12:
@cjandhyala can you please take a look?

andy31415, 2022-07-12:
@morahman2022 - please provide the commands that are included in `TC-BINFO-2.2` in this bug report. I opened the spreadsheet and still have a hard time finding the information.

andy31415, 2022-07-12:
Cert blocker review: `clean shutdown` is indeed not generally supported for devices. We may need to do one of:
   - figure out a trigger of such shutdown (e.g. test trigger)
   - can we even ask for graceful shutdown

for 2b, human log review is generally subjective. Could we automate this or can the test case be reviewed for correctness? We need for details in this issue to make an assesment.

DUT logs are generally not acceptable for test cases - firmware on embedded devices may not provide logs and it should still be able to pass cert. Some small devices disable logging due to flash size constraints.

@cjandhyala DUT logs cannot be used for certification, what can we do here?

cjandhyala, 2022-07-13:
@morahman2022 , as per the verification steps document, this test case is not ready for execution. Pls do not execute until the test case is ready. closing the issue for now.

closed 2022-07-13 by cjandhyala

lisaaaas, 2023-04-06:
Hello, I want to know  TC-BINFO-2.2 step 2 what shut down the DUT means. I have the same result as you, but I have not solved this problem at present

stefano055415, 2023-05-26:
@lisaaaas @cjandhyala @bzbarsky-apple 

I too have the same problem, I don't understand how to execute the shutdown command

is there a way via chip-tool?

bzbarsky-apple, 2023-05-26:
> is there a way via chip-tool?

Not for a real device.  Has to be done manually.