project-chip/connectedhomeip #20294: [Pre-SVE] [TC-CNET-4.6] Chip-Tool Verification Steps doesn't match Expected Outcome

AlvinHsiao, 2022-07-05:
Problem
Step 1 to Step 3 expects FAILSAFE_REQUIRED status code, but CHIP Tool Verification Steps show UNSUPPORTED_ACCESS.

bzbarsky-apple, 2022-07-12:
@AlvinHsiao Which SHA are you using?  Does this SHA include the fixes from #19964?

bzbarsky-apple, 2022-07-12:
The specific commands are, to be executed without a fail-safe armed:

```
chip-tool networkcommissioning add-or-update-thread-network 1232034768527434274  51 0

chip-tool networkcommissioning remove-network 1232034768527434274 51 0

chip-tool networkcommissioning connect-network 1232034768527434274 51 0
```

These all assume the device is commissioned as node id 51.

AlvinHsiao, 2022-07-14:
It seems fixed in the newest https://docs.google.com/spreadsheets/d/1h4d4J90KAxFqUL_cXZHyM3pHjHCEXzFITkQhN2KMqDY/edit#gid=509371791