project-chip/connectedhomeip #20309: [ota] Revise BDX timeout logic

Damian-Nordic, 2022-07-05:
#### Problem
OTA requestor does not use the [exchange response timeout](https://github.com/project-chip/connectedhomeip/blob/90f32a0f678074a9b73773c978eedfddeeab97f5/src/messaging/ExchangeContext.h#L185) for BDX messages. Instead, it uses a custom timeout mechanism which checks if the BDX downloader has made any progress within the last 5 minutes: https://github.com/project-chip/connectedhomeip/blob/90f32a0f678074a9b73773c978eedfddeeab97f5/src/app/clusters/ota-requestor/BDXDownloader.cpp#L39, https://github.com/project-chip/connectedhomeip/blob/90f32a0f678074a9b73773c978eedfddeeab97f5/src/app/clusters/ota-requestor/DefaultOTARequestor.cpp#L52.

It results in suboptimal behavior. If OTA requestor sends `BlockQuery` message and does not receive `Block` response within a few seconds, that is not reported to the BDX downloader. All that happes is that MRP silently gives up after a few retransmissions, and the BDX downloader cancels the OTA only after 5 minutes without making any subsequent attempts to download the next block. For all that time the BDX exchange is kept active which causes unnecessary battery drain in the case of Sleepy End Devices.

 #### Proposed Solution
Does anyone understand if the existing timeout mechanism is based on the spec or it's an implementation choice?
If it's the latter, maybe we could use the exchange response timeout mechanism and, say, make a few attempts of sending `BlockQuery` message until we give up?

carol-apple, 2022-07-18:
@Damian-Nordic It does look like OTA will call into the BDX time out code when the exchange times out. Would you care to experiment to see by calling `SetResponseTimeout`, we're actually catching the time out much sooner than the 5 min that is there currently?

carol-apple, 2022-07-19:
Spoke with @woody-apple and we'll take this for a v1 change but not into SVE branch.

Damian-Nordic, 2022-07-20:
@carol-apple I think given that the exchange timeout automatically closes the exchange, if the spec says 5 minutes is the minimum (thanks for the pointer, I missed this requirement!) we can't set the exchange timeout to anything less than that. But maybe the BDX code could try to re-send `BlockQuery` message after a minute or so.

I suspect the 5-minute timeout comes from the fact that the OTA provider may not have the file cached locally and must acquire it on the fly, but it many cases it seems it would be possible to increase the chances to complete the OTA transfer by repeating the query.

carol-apple, 2022-07-20:
Thanks @Damian-Nordic! I think the retry would be a great optimization that we can do in post v1. I don't believe as is, there's a clear API into BDX that allows us to retry so we'll need to put some thought into enabling that down the road. Moving this out of v1.

closed 2023-02-04 by stale[bot]