project-chip/connectedhomeip #20323: ScanNetworks (cmd=0x0, from cluster 0x31) during BLE provisioning is not working for Silabs platform

AlanLCollins, 2022-07-05:
#### Problem
during BLE provisioning, I am requesting the EFR32 lighting-app to perform ScanNetworks on Thread interface. I can see BLE activity happening in the serial logs, but no Thread beacon requests nor MLE discovery. 

The use case:
- The user has multiple Matter controllers at home. Only few of them support Thread interface. 
- The user acquires a matter accessory and starts provisioning over BLE. During the provisioning we want to confirm the accessory device has line of sight to the Thread network, or else we request the user to get closer to the matter controller which is Thread-enabled. 

<what's wrong or missing, please include any applicable:
 * expected behavior:
     The EFR32 should returns a scanNetworks response with the information of all available networks in the 15.4 channels (from 11 to 26). 
 * actual behavior
     The EFR32 timeouts the failsafe timer. 
 * steps to reproduce

1. attestation is optional (request DAC, validate DAC, PAI, etc) 
2. PASE secure channel: (request CSR, send root certficate, opcert, )
3. Send cluster 0x31, command 0x0 

 * system configuration
TE9 tag

AlanLCollins, 2022-07-05:
[efr32_ligth-bulb_scanNetworks.txt](https://github.com/project-chip/connectedhomeip/files/9048161/efr32_ligth-bulb_scanNetworks.txt)

jmartinez-silabs, 2022-07-14:
This PR that was merge sunday https://github.com/project-chip/connectedhomeip/pull/19200
should fix the issue

closed 2022-09-01 by AlanLCollins