project-chip/connectedhomeip #20342: [crash] chip::Controller::TypedCommandCallback<chip::app::DataModel::NullObjectType>::OnResponse(chip::app::CommandSender*, chip::app::ConcreteCommandPath const&, chip::app::StatusIB const&, chip::TLV::TLVReader*)

aajain-com, 2022-07-06:
Seeing this crash reported during testing on iOS/tvOS. This is seen as recent as TE9 SHA `cfc35951be66a664a6efdadea56d1b8ea6e63e96`.

```
Thread 4 Crashed ↩::   Dispatch queue: com.zigbee.chip.framework.controller.workqueue
1   CHIP                          	       0x101dd46a4        chip::Controller::TypedCommandCallback<chip::app::DataModel::NullObjectType>::OnResponse(chip::app::CommandSender*, chip::app::ConcreteCommandPath const&, chip::app::StatusIB const&, chip::TLV::TLVReader*) + 160
2   CHIP                          	       0x101fb2800        chip::app::CommandSender::ProcessInvokeResponseIB(chip::app::InvokeResponseIB::Parser&) + 1812 (connectedhomeip/src/app/CommandSender.cpp:310)
3   CHIP                          	       0x101fb1fd8        chip::app::CommandSender::ProcessInvokeResponse(chip::System::PacketBufferHandle&&) + 648 (connectedhomeip/src/app/CommandSender.cpp:192)
4   CHIP                          	       0x101fb1af4        chip::app::CommandSender::OnMessageReceived(chip::Messaging::ExchangeContext*, chip::PayloadHeader const&, chip::System::PacketBufferHandle&&) + 344 (connectedhomeip/src/app/CommandSender.cpp:136)
5   CHIP                          	       0x1020ac4b8        chip::Messaging::ExchangeContext::HandleMessage(unsigned int, chip::PayloadHeader const&, chip::BitFlags<chip::Messaging::MessageFlagValues, unsigned int>, chip::System::PacketBufferHandle&&) + 640 (connectedhomeip/src/messaging/ExchangeContext.cpp:492)
6   CHIP                          	       0x1020aefdc        auto chip::Messaging::ExchangeManager::OnMessageReceived(chip::PacketHeader const&, chip::PayloadHeader const&, chip::SessionHandle const&, chip::SessionMessageDelegate::DuplicateMessage, chip::System::PacketBufferHandle&&)::$_0::operator()<chip::Messaging::ExchangeContext>(chip::Messaging::ExchangeContext*) const + 260 (connectedhomeip/src/messaging/ExchangeMgr.cpp:221)
7   CHIP                          	       0x1020aee6c        chip::internal::LambdaProxy<chip::Messaging::ExchangeContext, chip::Messaging::ExchangeManager::OnMessageReceived(chip::PacketHeader const&, chip::PayloadHeader const&, chip::SessionHandle const&, chip::SessionMessageDelegate::DuplicateMessage, chip::System::PacketBufferHandle&&)::$_0>::Call(void*, void*) + 36 (connectedhomeip/src/lib/support/Pool.h:126)
8   CHIP                          	       0x1020a65e0        chip::internal::StaticAllocatorBitmap::ForEachActiveObjectInner(void*, chip::Loop (*)(void*, void*)) + 280 (connectedhomeip/src/lib/support/Pool.cpp:95)
9   CHIP                          	       0x1020aeafc        chip::Loop chip::BitMapObjectPool<chip::Messaging::ExchangeContext, 8ul>::ForEachActiveObject<chip::Messaging::ExchangeManager::OnMessageReceived(chip::PacketHeader const&, chip::PayloadHeader const&, chip::SessionHandle const&, chip::SessionMessageDelegate::DuplicateMessage, chip::System::PacketBufferHandle&&)::$_0>(chip::Messaging::ExchangeManager::OnMessageReceived(chip::PacketHeader const&, chip::PayloadHeader const&, chip::SessionHandle const&, chip::SessionMessageDelegate::DuplicateMessage, chip::System::PacketBufferHandle&&)::$_0&&) + 88 (connectedhomeip/src/lib/support/Pool.h:263)
10  CHIP                          	       0x1020ae4e4        chip::Messaging::ExchangeManager::OnMessageReceived(chip::PacketHeader const&, chip::PayloadHeader const&, chip::SessionHandle const&, chip::SessionMessageDelegate::DuplicateMessage, chip::System::PacketBufferHandle&&) + 416 (connectedhomeip/src/messaging/ExchangeMgr.cpp:214)
11  CHIP                          	       0x1020d04a4        chip::SessionManager::SecureUnicastMessageDispatch(chip::PacketHeader const&, chip::Transport::PeerAddress const&, chip::System::PacketBufferHandle&&) + 1200 (connectedhomeip/src/transport/SessionManager.cpp:590)
12  CHIP                          	       0x1020cf8b0        chip::SessionManager::OnMessageReceived(chip::Transport::PeerAddress const&, chip::System::PacketBufferHandle&&) + 232 (connectedhomeip/src/transport/SessionManager.cpp:431)
13  CHIP                          	       0x1020d63ac        chip::TransportMgrBase::HandleMessageReceived(chip::Transport::PeerAddress const&, chip::System::PacketBufferHandle&&) + 224 (connectedhomeip/src/transport/TransportMgrBase.cpp:71)
14  CHIP                          	       0x10218ddc8        chip::Transport::Base::HandleMessageReceived(chip::Transport::PeerAddress const&, chip::System::PacketBufferHandle&&) + 108 (connectedhomeip/src/transport/raw/Base.h:102)
15  CHIP                          	       0x10218d9c4        chip::Transport::UDP::OnUdpReceive(chip::Inet::UDPEndPoint*, chip::System::PacketBufferHandle&&, chip::Inet::IPPacketInfo const*) + 168 (connectedhomeip/src/transport/raw/UDP.cpp:122)
16  CHIP                          	       0x10208ecf4        chip::Inet::UDPEndPointImplSockets::HandlePendingIO(chip::BitFlags<chip::System::SocketEventFlags, unsigned char>) + 1620 (connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:688)
```

mrjerryjohns, 2022-07-25:
@bzbarsky-apple would be able to help take a further look since it's on a Darwin-based platform?

woody-apple, 2022-08-30:
This is no longer showing up in crashes.

closed 2022-08-30 by woody-apple