project-chip/connectedhomeip #20343: [crash] chip::Controller::CommissioningWindowOpener::OnOpenCommissioningWindowSuccess(void*, chip::app::DataModel::NullObjectType const&)

aajain-com, 2022-07-06:
Seeing this crash reported during testing on iOS/tvOS. This is seen as recent as TE9 SHA `cfc35951be66a664a6efdadea56d1b8ea6e63e96 `.

```
Thread 7 Crashed ↩::   Dispatch queue: com.zigbee.chip.framework.controller.workqueue

0   CHIP                          	       0x105252da0 chip::Controller::CommissioningWindowOpener::OnOpenCommissioningWindowSuccess(void*, chip::app::DataModel::NullObjectType const&) + 84 (connectedhomeip/src/controller/CommissioningWindowOpener.cpp:212)
1   CHIP                          	       0x105252d78 chip::Controller::CommissioningWindowOpener::OnOpenCommissioningWindowSuccess(void*, chip::app::DataModel::NullObjectType const&) + 44 (connectedhomeip/src/controller/CommissioningWindowOpener.cpp:207)
2   CHIP                          	       0x104fd46a4 chip::Controller::TypedCommandCallback<chip::app::DataModel::NullObjectType>::OnResponse(chip::app::CommandSender*, chip::app::ConcreteCommandPath const&, chip::app::StatusIB const&, chip::TLV::TLVReader*)
3   CHIP                          	       0x1051b2800 chip::app::CommandSender::ProcessInvokeResponseIB(chip::app::InvokeResponseIB::Parser&) + 1812 (connectedhomeip/src/app/CommandSender.cpp:310)
4   CHIP                          	       0x1051b1fd8 chip::app::CommandSender::ProcessInvokeResponse(chip::System::PacketBufferHandle&&) + 648 (connectedhomeip/src/app/CommandSender.cpp:192)
5   CHIP                          	       0x1051b1af4 chip::app::CommandSender::OnMessageReceived(chip::Messaging::ExchangeContext*, chip::PayloadHeader const&, chip::System::PacketBufferHandle&&) + 344 (connectedhomeip/src/app/CommandSender.cpp:136)
6   CHIP                          	       0x1052ac4b8 chip::Messaging::ExchangeContext::HandleMessage(unsigned int, chip::PayloadHeader const&, chip::BitFlags<chip::Messaging::MessageFlagValues, unsigned int>, chip::System::PacketBufferHandle&&) + 640 (connectedhomeip/src/messaging/ExchangeContext.cpp:492)
7   CHIP                          	       0x1052aefdc auto chip::Messaging::ExchangeManager::OnMessageReceived(chip::PacketHeader const&, chip::PayloadHeader const&, chip::SessionHandle const&, chip::SessionMessageDelegate::DuplicateMessage, chip::System::PacketBufferHandle&&)::$_0::operator()<chip::Messaging::ExchangeContext>(chip::Messaging::ExchangeContext*) const + 260 (connectedhomeip/src/messaging/ExchangeMgr.cpp:221)
8   CHIP                          	       0x1052aee6c chip::internal::LambdaProxy<chip::Messaging::ExchangeContext, chip::Messaging::ExchangeManager::OnMessageReceived(chip::PacketHeader const&, chip::PayloadHeader const&, chip::SessionHandle const&, chip::SessionMessageDelegate::DuplicateMessage, chip::System::PacketBufferHandle&&)::$_0>::Call(void*, void*) + 36 (connectedhomeip/src/lib/support/Pool.h:126)
9   CHIP                          	       0x1052a65e0 chip::internal::StaticAllocatorBitmap::ForEachActiveObjectInner(void*, chip::Loop (*)(void*, void*)) + 280 (connectedhomeip/src/lib/support/Pool.cpp:95)
10  CHIP                          	       0x1052aeafc chip::Loop chip::BitMapObjectPool<chip::Messaging::ExchangeContext, 8ul>::ForEachActiveObject<chip::Messaging::ExchangeManager::OnMessageReceived(chip::PacketHeader const&, chip::PayloadHeader const&, chip::SessionHandle const&, chip::SessionMessageDelegate::DuplicateMessage, chip::System::PacketBufferHandle&&)::$_0>(chip::Messaging::ExchangeManager::OnMessageReceived(chip::PacketHeader const&, chip::PayloadHeader const&, chip::SessionHandle const&, chip::SessionMessageDelegate::DuplicateMessage, chip::System::PacketBufferHandle&&)::$_0&&) + 88 (connectedhomeip/src/lib/support/Pool.h:263)
11  CHIP                          	       0x1052ae4e4 chip::Messaging::ExchangeManager::OnMessageReceived(chip::PacketHeader const&, chip::PayloadHeader const&, chip::SessionHandle const&, chip::SessionMessageDelegate::DuplicateMessage, chip::System::PacketBufferHandle&&) + 416 (connectedhomeip/src/messaging/ExchangeMgr.cpp:214)
12  CHIP                          	       0x1052d04a4 chip::SessionManager::SecureUnicastMessageDispatch(chip::PacketHeader const&, chip::Transport::PeerAddress const&, chip::System::PacketBufferHandle&&) + 1200 (connectedhomeip/src/transport/SessionManager.cpp:590)
13  CHIP                          	       0x1052cf8b0 chip::SessionManager::OnMessageReceived(chip::Transport::PeerAddress const&, chip::System::PacketBufferHandle&&) + 232 (connectedhomeip/src/transport/SessionManager.cpp:431)
14  CHIP                          	       0x1052d63ac chip::TransportMgrBase::HandleMessageReceived(chip::Transport::PeerAddress const&, chip::System::PacketBufferHandle&&) + 224 (connectedhomeip/src/transport/TransportMgrBase.cpp:71)
15  CHIP                          	       0x10538ddc8 chip::Transport::Base::HandleMessageReceived(chip::Transport::PeerAddress const&, chip::System::PacketBufferHandle&&) + 108 (connectedhomeip/src/transport/raw/Base.h:102)
16  CHIP                          	       0x10538d9c4 chip::Transport::UDP::OnUdpReceive(chip::Inet::UDPEndPoint*, chip::System::PacketBufferHandle&&, chip::Inet::IPPacketInfo const*) + 168 (connectedhomeip/src/transport/raw/UDP.cpp:122)
17  CHIP                          	       0x10528ecf4 chip::Inet::UDPEndPointImplSockets::HandlePendingIO(chip::BitFlags<chip::System::SocketEventFlags, unsigned char>) + 1620 (connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:688)

```

bzbarsky-apple, 2022-07-14:
This is crashing on this line:
```
        self->mCommissioningWindowCallback->mCall(self->mCommissioningWindowCallback->mContext, self->mNodeId, CHIP_NO_ERROR,
                                                  self->mSetupPayload);
```
which suggests the `CommissioningWindowOpener` object is dead.

It's possible to get there on that SHA with malformed-enough data from the server, but I doubt that was happening....

woody-apple, 2022-08-30:
This is no longer showing up in crashes.

closed 2022-08-30 by woody-apple