project-chip/connectedhomeip #20344: [crash] chip::System::LayerImplSelect::CancelTimer(void (*)(chip::System::Layer*, void*), void*)

aajain-com, 2022-07-06:
Seeing this crash reported during testing on iOS/tvOS. This is seen as recent as TE9 SHA `cfc35951be66a664a6efdadea56d1b8ea6e63e96`.

```
2   CHIP                          	       0x1b4e8c8f0 chip::System::LayerImplSelect::CancelTimer(void (*)(chip::System::Layer*, void*), void*) + 156 (Sources/CHIPFramework/connectedhomeip/src/system/SystemLayerImplSelect.cpp:174)
3   CHIP                          	       0x1b4e75808 chip::Messaging::ReliableMessageMgr::StopTimer() + 84 (Sources/CHIPFramework/connectedhomeip/src/messaging/ReliableMessageMgr.cpp:368)
4   CHIP                          	       0x1b4e75ae0 chip::Messaging::ReliableMessageMgr::StartTimer() + 140 (Sources/CHIPFramework/connectedhomeip/src/messaging/ReliableMessageMgr.cpp:349)
5   CHIP                          	       0x1b4e760a0 chip::Messaging::ReliableMessageMgr::StartRetransmision(chip::Messaging::ReliableMessageMgr::RetransTableEntry*) + 208 (Sources/CHIPFramework/connectedhomeip/src/messaging/ReliableMessageMgr.cpp:230)
6   CHIP                          	       0x1b4e72910 chip::Messaging::ExchangeMessageDispatch::SendMessage(chip::SessionManager*, chip::SessionHandle const&, unsigned short, bool, chip::Messaging::ReliableMessageContext*, bool, chip::Protocols::Id, unsigned char, chip::System::PacketBufferHandle&&) + 1124 (Sources/CHIPFramework/connectedhomeip/src/messaging/ExchangeMessageDispatch.cpp:102)
7   CHIP                          	       0x1b4e70c18 chip::Messaging::ExchangeContext::SendMessage(chip::Protocols::Id, unsigned char, chip::System::PacketBufferHandle&&, chip::BitFlags<chip::Messaging::SendMessageFlags, unsigned short> const&) + 1060 (Sources/CHIPFramework/connectedhomeip/src/messaging/ExchangeContext.cpp:187)
8   CHIP                          	       0x1b4d9b060 chip::ChipError chip::Messaging::ExchangeContext::SendMessage<chip::Protocols::InteractionModel::MsgType, void>(chip::Protocols::InteractionModel::MsgType, chip::System::PacketBufferHandle&&, chip::BitFlags<chip::Messaging::SendMessageFlags, unsigned short> const&) + 116 (Sources/CHIPFramework/connectedhomeip/src/messaging/ExchangeContext.h:123)
9   CHIP                          	       0x1b4d9a680 chip::app::ReadClient::SendSubscribeRequest(chip::app::ReadPrepareParams&) + 2256 (Sources/CHIPFramework/connectedhomeip/src/app/ReadClient.cpp:920)
10  CHIP                          	       0x1b4d994a8 chip::app::ReadClient::SendRequest(chip::app::ReadPrepareParams&) + 104 (Sources/CHIPFramework/connectedhomeip/src/app/ReadClient.cpp:195)
11  CHIP                          	       0x1b48f84c8 -[CHIPDevice subscribeWithQueue:minInterval:maxInterval:params:cacheContainer:attributeReportHandler:eventReportHandler:errorHandler:subscriptionEstablished:] + 1952 (Sources/CHIPFramework/connectedhomeip/src/darwin/Framework/CHIP/CHIPDevice.mm:418

```

bzbarsky-apple, 2022-07-14:
Seems pretty similar to #20627.

woody-apple, 2022-08-30:
This is no longer showing up in crashes.

closed 2022-08-30 by woody-apple