project-chip/connectedhomeip #20369: Respond to CASE Sigma1 retransmission with Sigma2 message

kpark-apple, 2022-07-06:
#### Problem
A node doesn't respond to CASE Sigma1 retransmission with CASE sigma2 message.

Let's say there are node A and node B.
Node A sends CASE Sigma1 message to node B.
Node B receive the message and sends back CASE Sigma2 message.
However the message was lost in the network.
Node A retransmits CASE Sigma1 message.
In this case, Node B doesn't seem to send back Sigma2 message but instead just acknowledge the message.
Node A stops retransmission upon receiving some response.
Node A eventually times out and fails to establish CASE session with Node B.

In the following log (corresponding to Node B), message with counter 1985674004 is the CASE Sigma1 message.

```
2022-06-30 23:46:49.450315-0700 I (100321) chip[EM]: Received message of type 0x30 with protocolId (0, 0) and MessageCounter:1985674004 on exchange 15761r
2022-06-30 23:46:49.470315-0700 I (100341) chip[IN]: CASE Server received Sigma1 message. Starting handshake. EC 0x3ffca608
2022-06-30 23:46:49.480315-0700 I (100351) chip[IN]: CASE Server disabling CASE session setups
2022-06-30 23:46:49.490315-0700 I (100361) chip[IN]: Prepared unauthenticated message 0x3ffe8a44 to 0x0000000000000000 (0)  of type 0x10 and protocolId (0, 0) on exchange 15761r with MessageCounter:3053733051.
2022-06-30 23:46:49.500315-0700 I (100371) chip[IN]: Sending unauthenticated msg 0x3ffe8a44 with MessageCounter:3053733051 to 0x0000000000000000 at monotonic time: 00000000000184C2 msec
2022-06-30 23:46:49.520315-0700 I (100391) chip[SC]: Received Sigma1 msg
2022-06-30 23:46:49.530315-0700 I (100401) chip[SC]: CASE matched destination ID: fabricIndex 1, NodeID 0x00000000158AA618
2022-06-30 23:46:51.520315-0700 I (102391) chip[IN]: Prepared unauthenticated message 0x3ffca8bc to 0x0000000000000000 (0)  of type 0x31 and protocolId (0, 0) on exchange 15761r with MessageCounter:3053733052.
2022-06-30 23:46:51.530315-0700 I (102401) chip[IN]: Sending unauthenticated msg 0x3ffca8bc with MessageCounter:3053733052 to 0x0000000000000000 at monotonic time: 0000000000018CB4 msec
2022-06-30 23:46:51.550315-0700 I (102421) chip[SC]: Sent Sigma2 msg
2022-06-30 23:46:51.540315-0700 I (102411) ROUTE_HOOK: Received RIO
2022-06-30 23:46:51.550315-0700 I (102421) ROUTE_HOOK: prefix FDE7:CB40:7ED7:: lifetime 1800

2022-06-30 23:46:51.550315-0700 E (102421) chip[DL]: Long dispatch time: 2097 ms, for event type 3
2022-06-30 23:46:51.530315-0700 I (102401) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 0 12
2022-06-30 23:46:51.880315-0700 I (102751) chip[IN]: Sending unauthenticated msg 0x3ffca8bc with MessageCounter:3053733052 to 0x0000000000000000 at monotonic time: 0000000000018E0E msec
2022-06-30 23:46:52.210315-0700 I (103081) chip[IN]: Sending unauthenticated msg 0x3ffca8bc with MessageCounter:3053733052 to 0x0000000000000000 at monotonic time: 0000000000018F58 msec
2022-06-30 23:46:52.220315-0700 E (103091) chip[EM]: Crit-err 3000001 when sending CHIP MessageCounter:3053733052 on exchange 15761r, send tries: 2
2022-06-30 23:46:53.240315-0700 I (104111) chip[DL]: IP_EVENT_GOT_IP6
2022-06-30 23:46:53.240315-0700 I (104111) chip[DL]: IPv6 addr available. Ready on WIFI_STA_DEF interface: fd15:fed4:2265:4925:baf0:09ff:fec4:8ff0
2022-06-30 23:46:53.250315-0700 I (104121) app-devicecallbacks: Current free heap: 59836

2022-06-30 23:46:53.250315-0700 I (104121) chip[DIS]: Updating services using commissioning mode 0
2022-06-30 23:46:53.260315-0700 I (104131) chip[DIS]: CHIP minimal mDNS started advertising.
2022-06-30 23:46:53.280315-0700 I (104151) chip[DIS]: Advertise operational node 6BD6C52C30FD8FFC-00000000158AA618
2022-06-30 23:46:53.280315-0700 I (104151) chip[DIS]: CHIP minimal mDNS configured as 'Operational device'.
2022-06-30 23:46:53.300315-0700 I (104171) chip[DIS]: mDNS service published: _matter._tcp
2022-06-30 23:46:53.300315-0700 I (104171) ROUTE_HOOK: Hook already installed on netif, skip...
2022-06-30 23:46:53.310315-0700 I (104181) app-devicecallbacks: Current free heap: 56872

2022-06-30 23:46:54.080315-0700 I (104951) NimBLE: GATT procedure initiated: notify;
2022-06-30 23:46:54.080315-0700 I (104951) NimBLE: att_handle=14

2022-06-30 23:46:54.080315-0700 I (104951) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 0) status= 0
2022-06-30 23:46:54.670315-0700 I (105541) chip[EM]: Received message of type 0x30 with protocolId (0, 0) and MessageCounter:1985674004 on exchange 15761r
2022-06-30 23:46:54.680315-0700 I (105551) chip[IN]: Prepared unauthenticated message 0x3ffe8a14 to 0x0000000000000000 (0)  of type 0x10 and protocolId (0, 0) on exchange 15761r with MessageCounter:3053733053.
2022-06-30 23:46:54.700315-0700 I (105571) chip[IN]: Sending unauthenticated msg 0x3ffe8a14 with MessageCounter:3053733053 to 0x0000000000000000 at monotonic time: 0000000000019913 msec
```

 #### Proposed Solution
Fix Matter stack either to send back CASE Sigma2 message for a received retransmission of CASE Sigma1 message or to retransmit CASE Sigma2 message if CASE Sigma3 message is not received. Otherwise, loss of one UDP packet may cause session establishment failure.

bzbarsky-apple, 2022-07-12:
> Node B receive the message and sends back CASE Sigma2 message.
> However the message was lost in the network.

In this case Node B retransmits the Sigma2 until it's acked or we reach the MRP timeout, no?

closed 2022-07-12 by bzbarsky-apple

kpark-apple, 2022-07-12:
> > Node B receive the message and sends back CASE Sigma2 message.
> > However the message was lost in the network.
> 
> In this case Node B retransmits the Sigma2 until it's acked or we reach the MRP timeout, no?

No, Node B didn't retransmit. Maybe there is a bug there.

kpark-apple, 2022-07-12:
The following was the controller side log:

```
2022-06-30 23:46:49.353226-0700 0x21f0b homed: (CHIP) [com.zigbee.chip:all] 🔵 [1656658009365] [839:139019] CHIP: [IN] Prepared unauthenticated message 0x113814c68 to 0x0000000000000000 (0)  of type 0x30 and protocolId (0, 0) on exchange 15761i with MessageCounter:1985674004.
2022-06-30 23:46:49.353243-0700 0x21f0b homed: (CHIP) [com.zigbee.chip:all] 🔵 [1656658009365] [839:139019] CHIP: [IN] Sending unauthenticated msg 0x113814c68 with MessageCounter:1985674004 to 0x0000000000000000 at monotonic time: 000000000178658E msec
2022-06-30 23:46:49.353561-0700 0x21f0b homed: (CHIP) [com.zigbee.chip:all] 🔵 [1656658009366] [839:139019] CHIP: [SC] Sent Sigma1 msg
2022-06-30 23:46:54.618725-0700 0x21f0b homed: (CHIP) [com.zigbee.chip:all] 🟢 [1656658014631] [839:139019] CHIP: [EM] Retransmitting MessageCounter:1985674004 on exchange 15761i Send Cnt 1
2022-06-30 23:46:54.757000-0700 0x21dfb homed: (CHIP) [com.zigbee.chip:all] 🔵 [1656658014769] [839:138747] CHIP: [EM] Received message of type 0x10 with protocolId (0, 0) and MessageCounter:3053733053 on exchange 15761i
2022-06-30 23:46:54.757011-0700 0x21dfb homed: (CHIP) [com.zigbee.chip:all] 🟢 [1656658014769] [839:138747] CHIP: [EM] Found matching exchange: 15761i, Delegate: 0x106812250
2022-06-30 23:47:44.354694-0700 0x21f87 homed: (CHIP) [com.zigbee.chip:all] 🔴 [1656658064367] [839:139143] CHIP: [SC] CASESession timed out while waiting for a response from the peer. Current state was 1
```

bzbarsky-apple, 2022-07-12:
So these bits from the device log:
```
2022-06-30 23:46:51.530315-0700 I (102401) chip[IN]: Sending unauthenticated msg 0x3ffca8bc with MessageCounter:3053733052 to 0x0000000000000000 at monotonic time: 0000000000018CB4 msec
2022-06-30 23:46:51.550315-0700 I (102421) chip[SC]: Sent Sigma2 msg
...
2022-06-30 23:46:51.880315-0700 I (102751) chip[IN]: Sending unauthenticated msg 0x3ffca8bc with MessageCounter:3053733052 to 0x0000000000000000 at monotonic time: 0000000000018E0E msec
2022-06-30 23:46:52.210315-0700 I (103081) chip[IN]: Sending unauthenticated msg 0x3ffca8bc with MessageCounter:3053733052 to 0x0000000000000000 at monotonic time: 0000000000018F58 msec
2022-06-30 23:46:52.220315-0700 E (103091) chip[EM]: Crit-err 3000001 when sending CHIP MessageCounter:3053733052 on exchange 15761r, send tries: 2
```
That does in fact show the Sigma2 being retransmitted.  One of the retransmits fails synchronously causing retransmits to stop.

The error there, `3000001` is an underlying LwIP error.  Specifically, `ERR_MEM`, I believe.

Kxuan, 2022-08-03:
@kpark-apple  @bzbarsky-apple 

We found the same problem in our auto tests. The log on Node B shows that it sent Sigma2 message 3 times, and in the last times, lwIP returns an error code 3000001. But Node A does not receive any Sigma2 message.

The log messages shows that it should not have a memory insufficient problem in the time. So I think `ERR_MEM` is not the root reason.

So maybe we can reopen this issue, and focus on why sending retransmitted messages cause lwIP returns `ERR_MEM`.

```
2022-08-02 22:08:36.886334 : b'\x1b[0;32mI (60910) chip[IN]: Prepared unauthenticated message 0x3ffc6ae4 to 0x0000000000000000 (0)  of type 0x31 and protocolId (0, 0) on exchange 31470r with MessageCounter:8665462.\x1b[0m\r\n'
2022-08-02 22:08:36.902290 : b'\x1b[0;32mI (60930) chip[IN]: Sending unauthenticated msg 0x3ffc6ae4 with MessageCounter:8665462 to 0x0000000000000000 at monotonic time: 000000000000EB04 msec\x1b[0m\r\n'
2022-08-02 22:08:36.903256 : b'\x1b[0;32mI (60940) chip[SC]: Sent Sigma2 msg\x1b[0m\r\n'
2022-08-02 22:08:36.917256 : b'\x1b[0;31mE (60940) chip[DL]: Long dispatch time: 1007 ms, for event type 3\x1b[0m\r\n'
2022-08-02 22:08:36.918248 : b'\x1b[0;32mI (60940) chip[DL]: IP_EVENT_GOT_IP6\x1b[0m\r\n'
2022-08-02 22:08:36.933206 : b'\x1b[0;32mI (60960) chip[DL]: IPv6 addr available. Ready on WIFI_STA_DEF interface: fe80:0000:0000:0000:9af4:abff:fe9a:4f44\x1b[0m\r\n'
2022-08-02 22:08:36.934210 : b'\x1b[0;32mI (60970) app-devicecallbacks: Current free heap: 94456\r\n'
2022-08-02 22:08:36.934210 : b'\x1b[0m\r\n'
2022-08-02 22:08:36.936168 : b'\x1b[0;32mI (60970) chip[DIS]: Updating services using commissioning mode 0\x1b[0m\r\n'
2022-08-02 22:08:36.949163 : b'\x1b[0;32mI (60970) chip[DIS]: CHIP minimal mDNS started advertising.\x1b[0m\r\n'
2022-08-02 22:08:36.965156 : b'\x1b[0;32mI (60990) chip[DIS]: Advertise operational node 4820932C08EB5A10-062A538BC8ECCA00\x1b[0m\r\n'
2022-08-02 22:08:36.967085 : b"\x1b[0;32mI (61000) chip[DIS]: CHIP minimal mDNS configured as 'Operational device'.\x1b[0m\r\n"
2022-08-02 22:08:36.981167 : b'\x1b[0;32mI (61010) chip[DIS]: mDNS service published: _matter._tcp\x1b[0m\r\n'
2022-08-02 22:08:36.983165 : b'\x1b[0;32mI (61010) app-devicecallbacks: Current free heap: 94316\r\n'
2022-08-02 22:08:36.983165 : b'\x1b[0m\r\n'
2022-08-02 22:08:37.173972 : b'\x1b[0;32mI (61210) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 0 12\x1b[0m\r\n'
2022-08-02 22:08:37.270616 : b'\x1b[0;32mI (61290) chip[IN]: Sending unauthenticated msg 0x3ffc6ae4 with MessageCounter:8665462 to 0x0000000000000000 at monotonic time: 000000000000EC71 msec\x1b[0m\r\n'
2022-08-02 22:08:37.589442 : b'\x1b[0;32mI (61610) chip[IN]: Sending unauthenticated msg 0x3ffc6ae4 with MessageCounter:8665462 to 0x0000000000000000 at monotonic time: 000000000000EDAF msec\x1b[0m\r\n'
2022-08-02 22:08:37.593421 : b'\x1b[0;31mE (61610) chip[EM]: Crit-err 3000001 when sending CHIP MessageCounter:8665462 on exchange 31470r, send tries: 2\x1b[0m\r\n'
```

bzbarsky-apple, 2022-08-03:
@Kxuan If you have a way to reproduce, I would suggest filing a new issue and attaching (not pasting) complete logs.  That might help figure out what's going on here....

asakhart, 2022-08-08:
For posterity, was there a PR that resulted in this issue being closed?

kpark-apple, 2022-08-08:
The issue was closed because the cause wasn't the lack of Sigma2 message retransmissions from the Matter stack. The proposed solution in the description is consequently incorrect.