project-chip/connectedhomeip #20384: [TC-SWTCH] - Switch cluster test case needs an reference app to modify and verify the state of the the switch position.

somu1710, 2022-07-06:
**Problem description:**

To verify the latching and Momentary switch state as explained in the test case TC-SWTCH-3.2, If there is an reference app to modify the state of the switch will be helpful to verify this simulated functionality.

currently test case expects to change the state of the switch from 0 to 1 and vice versa and expects to read the attribute. As of now there is no provision to change the change the state of the switch. Hence requesting for a reference app to verify this functionality.

For test case TC-SWTCH-2.2 - Also needs an reference app to change the switch position and returns to the existing position.
**For ex**: The test steps needs to operate the switch and release the switch by the operator.

For your reference here I attached the Draft TE9 document link:
https://docs.google.com/spreadsheets/d/1h4d4J90KAxFqUL_cXZHyM3pHjHCEXzFITkQhN2KMqDY/edit?pli=1#gid=1872755018

**Test plan link:**
https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/switch.adoc

somu1710, 2022-07-06:
@yufengwangca  could you please take a look at this issue.

markus-becker-tridonic-com, 2022-07-13:
all-clusters-app can do this by sending SIG* to the Linux process:
https://github.com/project-chip/connectedhomeip/blob/master/examples/all-clusters-app/linux/main-common.cpp#L284-L311

However, it might not be completely correct (https://github.com/project-chip/connectedhomeip/issues/19923).

markus-becker-tridonic-com, 2022-07-20:
For your reference, these are the commands I have been using for testing a MS switch:

```
./chip-tool interactive start
switch read current-position 1234478 2
switch subscribe-event initial-press 1 100 1234478 2 --is-urgent true 
switch subscribe-event short-release 1 100 1234478 2 --is-urgent true --keepSubscriptions true
```

cjandhyala, 2022-07-20:
@markus-becker-tridonic-com question is how are to triggering those events on reference apps. I believe your DUT generate those events by pressing keys on your DUT ?

franck-apple, 2022-07-21:
assigning to @markus-becker-tridonic-com and @somu1710 for feedback.

markus-becker-tridonic-com, 2022-07-22:
> @markus-becker-tridonic-com question is how are to triggering those events on reference apps. I believe your DUT generate those events by pressing keys on your DUT ?

indeed on our DUT we do that.

Issue https://github.com/project-chip/connectedhomeip/issues/19923 was created before https://github.com/project-chip/connectedhomeip/pull/20560. Have not checked what the current state of switch support in all-clusters-app is.

It seems https://github.com/project-chip/connectedhomeip/pull/20560 has instructions to trigger events. The events are listed in https://github.com/project-chip/connectedhomeip/blob/master/examples/all-clusters-app/linux/AllClustersCommandDelegate.cpp#L42-L118.

woody-apple, 2022-07-22:
Cert Blocker Review: @markus-becker-tridonic-com Is this a real DUT failing? This current issue is tracking reference apps only. 

Can we better understand the setup that we're testing here?

yufengwangca, 2022-07-24:
Per talking with @somu1710, #21144 add the capability to adjust the state of the switch using json payload

markus-becker-tridonic-com, 2022-07-25:
@woody-apple i believe our DUT would be passing the TC-SWTCH testcases. If not (depends on changes to test-plans/scripts), we would be able to accomodate changes for our DUT.

I believe the open topic is that the all-clusters-app (not real DUT) does not offer the right features to test controllers.

PR #21144 (2022-07-25): [TC-SWTCH]: Add capability to adjust the state of the the switch position
#### Problem
What is being fixed?  Examples:
* To verify the latching and Momentary switch state as explained in the test case TC-SWTCH-3.2, we need a way to modify the state of the switch will be helpful to verify this simulated functionality.

* currently test case expects to change the state of the switch from 0 to 1 and vice versa and expects to read the attribute. As of now there is no provision to change the change the state of the switch. Hence requesting for a way to verify this functionality.

* For test case TC [... cut]
PR #21184 (2022-07-26): [TC-SWTCH]: Add capability to adjust the state of the the switch position
#### Problem
What is being fixed?  Examples:
* To verify the latching and Momentary switch state as explained in the test case TC-SWTCH-3.2, we need a way to modify the state of the switch will be helpful to verify this simulated functionality.

* currently test case expects to change the state of the switch from 0 to 1 and vice versa and expects to read the attribute. As of now there is no provision to change the change the state of the switch. Hence requesting for a way to verify this functionality.

* For test case TC [... cut]
closed 2022-07-25 by woody-apple