project-chip/connectedhomeip #20385: QR code generator: add support for manufacturer specific tags

leorozendaal, 2022-07-06:
#### Problem
Current QR code generator (according to @liamgoneya) only supports the Matter-defined TLV tags  (#20226 for #20231) .
This prevents us from making a test whether commissioners can handle a QR with a (large) manufacturer specific blob of TLV data.

 #### Proposed Solution
Add support for generating, possibly with the client/command line providing the entire blob (tag, length, value, data) which is woven into the appropriate place in the QR code data.

@bzbarsky-apple

liamgonyea, 2022-07-06:
#### Example of the QR code generator's current lack of support:

The <Payload in Base-38 format for 1000 bytes of TLV data> could be +
0x15 0x31 0x82 0xe8 0x03 <1000 bytes of random data> 0x18 +
(using tag type 0x31='octet string, 2-octet length', vendor tag 0x82, length 0x03e8)

```
15
31
82
e8
03
31323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930
18
```

 ```
16-MacBook-Pro:connectedhomeip liam$ ./out/chip-tool/chip-tool payload generate-qrcode --existing-payload MT:-24J0AFN00KA0648G00  --tlvBytes hex:153182e8033132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393018
[1657131625253] [13981:15903653] CHIP: [TOO] Unexpected type for vendor data: 16
[1657131625253] [13981:15903653] CHIP: [TOO] Unable to populate payload TLV: ../../examples/chip-tool/commands/payload/SetupPayloadGenerateCommand.cpp:189: Error 0x00000026
[1657131625253] [13981:15903653] CHIP: [TOO] Run command failure: ../../examples/chip-tool/commands/payload/SetupPayloadGenerateCommand.cpp:189: Error 0x00000026
16-MacBook-Pro:connectedhomeip liam$ 
```

### Test Setup
```
commit b245151d694b0a86ef0c293fe61ae138ac5d8cd5 (HEAD -> TLVdata)
Author: Boris Zbarsky <bzbarsky@apple.com>
Date:   Fri Jul 1 12:24:35 2022 -0700

    Allow chip-tool to generate onboarding payloads with extra TLV.
    
    Specific changes:
    
    1) Clarify the naming of some of the chip-tool payload arguments and add
       documentation.
    2) Fix chip-tool handling of the existing-payload argument so that we error out
       on invalid existing payloads instead of silently pressing on.
    3) Add a way to pass in the TLV-encoded extra bytes to be added to a payload.
       Unfortunately, only tags that our SetupPayload knows about right now are
       supported.
    4) Add a function on QRCodeSetupPayloadGenerator that allows generating a code
       without having to guess at how much space the TLV will take up.
    5) Add tests for that new function.
    
    Fixes https://github.com/project-chip/connectedhomeip/issues/20226
```

bzbarsky-apple, 2022-07-12:
> Current QR code generator (according to @liamgonyea) only supports the Matter-defined TLV tags

We support vendor tags, but the only value types we support are char string and signed integer, because those are the only things supported by the `SetupPayload` type.  The above is trying to use octet string....

It's pretty easy to change the chip-tool bits here if `SetupPayload` is fixed to actually support other types.  I don't know why it does not do that now....

woody-apple, 2022-07-13:
Cert Blocker Review: Pending cert blocker, given resolution of this issue - is this required for 1.0 or 1.X?: https://github.com/CHIP-Specifications/chip-test-plans/issues/1659

liamgonyea, 2022-07-19:
@leorozendaal Do you have input on whether this is V1.0 or V1.x?

woody-apple, 2022-07-19:
Cert Blocker Review: For v1, no tooling is planned, we recommend using a hard coded payload. @bzbarsky-apple also has a recommendation as well for some alternative encoding that may work.

bzbarsky-apple, 2022-07-20:
> Content can be char string instead of octet string

In that case, the existing generator will work just fine.  Using 0x2d instead of 0x31 for the control tag for the string in the TLV above (2-octet-length UTF-8 string instead of byte string):
```
chip-tool payload generate-qrcode --existing-payload MT:-24J0AFN00KA0648G00  --tlvBytes hex:152d82e8033132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393018
```
gives me:
```
MT:-24J0AFN00KA064IJ3P0JFQB7TZZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1
```

liamgonyea, 2022-07-20:
@bzbarsky-apple @leorozendaal that works for me to generate the random 1000 bytes of TLV data into a QR code, however `chip-tool` does not seem to support commissioning with this QR code, which it would be nice if `chip-tool` could be an example for here

> 16-MacBook-Pro:debug liam$ ./chip-tool pairing code 1 MT:-24J0AFN00KA064IJ3P0JFQB7TZZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1
[1658359318193] [3713:4181405] CHIP: [DL] _Init
[1658359318193] [3713:4181405] CHIP: [IN] UDP::Init bind&listen port=0
[1658359318193] [3713:4181405] CHIP: [IN] UDP::Init bound to port=60923
[1658359318193] [3713:4181405] CHIP: [IN] BLEBase::Init - setting/overriding transport
[1658359318193] [3713:4181405] CHIP: [IN] TransportMgr initialized
[1658359318193] [3713:4181405] CHIP: [FP] Initializing FabricTable from persistent storage
[1658359318194] [3713:4181405] CHIP: [TS] Last Known Good Time: 2022-07-09T15:20:22
[1658359318194] [3713:4181405] CHIP: [FP] Fabric index 0x1 was retrieved from storage. Compressed FabricId 0xD7479342D7C73055, FabricId 0x0000000000000001, NodeId 0x000000000001B669, VendorId 0xFFF1
[1658359318194] [3713:4181405] CHIP: [FP] Fabric index 0x2 was retrieved from storage. Compressed FabricId 0x70E58FA9009AE023, FabricId 0x0000000000000002, NodeId 0x000000000001B669, VendorId 0xFFF1
[1658359318194] [3713:4181405] CHIP: [FP] Fabric index 0x3 was retrieved from storage. Compressed FabricId 0xDE008D6D9E9DAFD8, FabricId 0x0000000000000003, NodeId 0x000000000001B669, VendorId 0xFFF1
[1658359318196] [3713:4181405] CHIP: [ZCL] Using ZAP configuration...
[1658359318196] [3713:4181405] CHIP: [CTL] System State Initialized...
[1658359318196] [3713:4181405] CHIP: [CTL] Stopping commissioning discovery over DNS-SD
[1658359318196] [3713:4181405] CHIP: [CTL] Setting attestation nonce to random value
[1658359318196] [3713:4181405] CHIP: [CTL] Setting CSR nonce to random value
[1658359318196] [3713:4181405] CHIP: [CTL] Stopping commissioning discovery over DNS-SD
[1658359318196] [3713:4181405] CHIP: [CTL] Setting attestation nonce to random value
[1658359318196] [3713:4181405] CHIP: [CTL] Setting CSR nonce to random value
[1658359318199] [3713:4181405] CHIP: [CTL] Generating NOC
[1658359318202] [3713:4181405] CHIP: [FP] Validating NOC chain
[1658359318212] [3713:4181405] CHIP: [FP] NOC chain validation successful
[1658359318212] [3713:4181405] CHIP: [FP] Updated fabric at index: 0x1, Node ID: 0x000000000001B669
[1658359318212] [3713:4181405] CHIP: [TS] Last Known Good Time: 2022-07-09T15:20:22
[1658359318212] [3713:4181405] CHIP: [TS] New proposed Last Known Good Time: 2021-01-01T00:00:00
[1658359318212] [3713:4181405] CHIP: [TS] Retaining current Last Known Good Time
[1658359318214] [3713:4181405] CHIP: [FP] Metadata for Fabric 0x1 persisted to storage.
[1658359318215] [3713:4181405] CHIP: [TS] Committing Last Known Good Time to storage: 2022-07-09T15:20:22
[1658359318216] [3713:4181405] CHIP: [CTL] Joined the fabric at index 1. Compressed fabric ID is: 0x0000000000000000
[1658359318216] [3713:4181405] CHIP: [CTL] Stopping commissioning discovery over DNS-SD
[1658359318216] [3713:4181405] CHIP: [CTL] Setting attestation nonce to random value
[1658359318216] [3713:4181405] CHIP: [CTL] Setting CSR nonce to random value
[1658359318219] [3713:4181405] CHIP: [CTL] Generating NOC
[1658359318222] [3713:4181405] CHIP: [FP] Validating NOC chain
[1658359318232] [3713:4181405] CHIP: [FP] NOC chain validation successful
[1658359318232] [3713:4181405] CHIP: [FP] Updated fabric at index: 0x2, Node ID: 0x000000000001B669
[1658359318232] [3713:4181405] CHIP: [TS] Retaining current Last Known Good Time
[1658359318234] [3713:4181405] CHIP: [FP] Metadata for Fabric 0x2 persisted to storage.
[1658359318235] [3713:4181405] CHIP: [TS] Committing Last Known Good Time to storage: 2022-07-09T15:20:22
[... cut ...]
[1658359318280] [3713:4181405] CHIP: [DMG] All ReadHandler-s are clean, clear GlobalDirtySet
[1658359318280] [3713:4181405] CHIP: [BLE] CancelConnection
[1658359318280] [3713:4181405] CHIP: [DL] Inet Layer shutdown
[1658359318280] [3713:4181405] CHIP: [DL] BLE shutdown
[1658359318280] [3713:4181405] CHIP: [DL] System Layer shutdown
[1658359318280] [3713:4181405] CHIP: [TOO] Run command failure: ../../third_party/connectedhomeip/src/setup_payload/Base38Decode.cpp:155: CHIP Error 0x0000002F: Invalid argument

bzbarsky-apple, 2022-07-21:
Ah, so... The generate-qrcode command uses Matter logging to print the resulting code, but Matter logging has a fixed-size buffer and can only log so much at once, so we are getting a truncated QR code.

The right QR code is:
```
MT:-24J0AFN00KA064IJ3P0JFQB7TZZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T1VFSK1S3DO1ZTZR1UNMJ1DK5N1K8SQ1RYCU1--ZL15PKP1CD5T11UXS0
```
and with that I run into bugs in our base38 decoder when the input is too long!  Will create a PR for that.

bzbarsky-apple, 2022-07-21:
https://github.com/project-chip/connectedhomeip/pull/21037 fixes the parsing bug.

bzbarsky-apple, 2022-07-22:
@liamgonyea Are you good here now?

liamgonyea, 2022-07-23:
@bzbarsky-apple Thanks for the PR, the pairing using [the code above ](https://github.com/project-chip/connectedhomeip/issues/20385#issuecomment-1190986850) works with chip-tool now, all good, closing the ticket as it's resolved

bzbarsky-apple, 2022-08-12:
Filed https://github.com/project-chip/connectedhomeip/issues/21843 to track the missing bits in SetupPayload, and this is resolved as originally filed.

closed 2022-08-12 by bzbarsky-apple