project-chip/connectedhomeip #20470: [ESP32] cannot find Discriminator: 3840 on all-cluster-app via chip-device-ctrl(python)

tatsutaigu, 2022-07-08:
#### Problem
 * expected behavior

`ble-scan` on `chip-device-ctrl` can find discriminator:3840.

> chip-device-ctrl > ble-scan
> 2022-07-08 09:28:10,520 ChipBLEMgr   INFO     scanning started
> 2022-07-08 09:28:11,044 ChipBLEMgr   INFO     Name            = None
> 2022-07-08 09:28:11,045 ChipBLEMgr   INFO     ID              = 12a18652-bbd8-41e5-ac1c-35067f35d0cb
> 2022-07-08 09:28:11,047 ChipBLEMgr   INFO     RSSI            = -59
> 2022-07-08 09:28:11,048 ChipBLEMgr   INFO     Address         = 7E:1E:1E:79:4F:F7
> 2022-07-08 09:28:11,053 ChipBLEMgr   INFO     Pairing State   = 0
> 2022-07-08 09:28:11,053 ChipBLEMgr   INFO     Discriminator   = 3840
> 2022-07-08 09:28:11,054 ChipBLEMgr   INFO     Vendor Id       = 9050
> 2022-07-08 09:28:11,054 ChipBLEMgr   INFO     Product Id      = 17729
> 2022-07-08 09:28:11,077 ChipBLEMgr   INFO     Adv UUID        = 0000fff6-0000-1000-8000-00805f9b34fb
> 2022-07-08 09:28:11,082 ChipBLEMgr   INFO     Adv Data        = 00000f5a234145
> 2022-07-08 09:28:11,083 ChipBLEMgr   INFO     
> 2022-07-08 09:28:21,198 ChipBLEMgr   INFO     scanning stopped
> chip-device-ctrl > 

 * actual behavior

`ble-scan` cannot detect discriminator.
`chip-device-ctrl` on master branch does not work.
therefore, I use it on [test_event_4_2021_07_06](https://github.com/project-chip/connectedhomeip/tree/test_event_4_2021_07_06/scripts/build_python.sh).

* steps to reproduce
* system configuration
Board : ESP32-C3-DevKitM-1

    * connectedhomeip version

```
$ git log -1
commit 126f6b2915fc26ee17342e1dc80f96f031938a6c (HEAD -> master, origin/master, origin/HEAD)
Author: Stefan Agner <stefan@agner.ch>
Date:   Wed Jul 6 15:04:20 2022 +0200

   [python] Move string encoding to ChipDeviceCtrl.py (#20306)
    
   * Encode strings in ChipDeviceCtrl.py
    
   * Fix Python tests
```

    * ESP-IDF version

```
$ git log -1
commit 1329b19fe494500aeb79d19b27cfd99b40c37aec (HEAD, tag: v4.4.1)
Author: Michael (XIAO Xufeng) <xiaoxufeng@espressif.com>
Date:   Sat Apr 16 09:43:14 2022 +0800

    versions: Update version to 4.4.1
```
* screenshots, images, logs, etc.

> chip-device-ctrl > ble-scan
> 2022-07-08 08:08:54,980 ChipBLEMgr   INFO     scanning started
> 2022-07-08 08:08:55,225 ChipBLEMgr   INFO     Name            = None
> 2022-07-08 08:08:55,226 ChipBLEMgr   INFO     ID              = 4ab2afe7-351f-4c98-a74b-a9c565ca9dba
> 2022-07-08 08:08:55,228 ChipBLEMgr   INFO     RSSI            = -32
> 2022-07-08 08:08:55,230 ChipBLEMgr   INFO     Address         = C8:E8:F5:91:36:6F
> 2022-07-08 08:08:55,238 ChipBLEMgr   INFO     Adv UUID        = 0000fff6-0000-1000-8000-00805f9b34fb
> 2022-07-08 08:08:55,240 ChipBLEMgr   INFO     Adv Data        = 00000ff1ff008000
> 2022-07-08 08:08:55,240 ChipBLEMgr   INFO     
> 2022-07-08 08:09:05,171 ChipBLEMgr   INFO     scanning stopped
> chip-device-ctrl >

 #### Proposed Solution
<suggested fix, suggested enhancement>

dhrishi, 2022-07-08:
>2022-07-08 08:08:55,240 ChipBLEMgr INFO Adv Data = 00000ff1ff008000

HI @tatsutaigu The "Adv Data" that you see above has the discriminator: 0xF00 (3840), VID: 0xFFF1 as well as PID: 0x8001 in the format as required by the specification.

tatsutaigu, 2022-07-11:
Hi, @dhrishi 
Thank you for your reply.

I use `chip-tool` and it can find ESP32.
However, `chip-tool` was stopped as Aborted (core dumped).

Below is log.

> $ ./out/debug/chip-tool pairing ble-wifi 12344321 $SSID $PASSWORD 20202021 3840
> [1657513261.977971][3087:3087] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_kvs
> [1657513261.978251][3087:3087] CHIP:DL: writing settings to file (/tmp/chip_kvs-pGJSPA)
> [1657513261.978420][3087:3087] CHIP:DL: renamed tmp file to file (/tmp/chip_kvs)
> [1657513261.978737][3087:3087] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_factory.ini
> [1657513261.978846][3087:3087] CHIP:DL: writing settings to file (/tmp/chip_factory.ini-nGbQjC)
> [1657513261.978979][3087:3087] CHIP:DL: renamed tmp file to file (/tmp/chip_factory.ini)
> [1657513261.979053][3087:3087] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_config.ini
> [1657513261.979155][3087:3087] CHIP:DL: writing settings to file (/tmp/chip_config.ini-QyrBpC)
> [1657513261.979263][3087:3087] CHIP:DL: renamed tmp file to file (/tmp/chip_config.ini)
> [1657513261.979332][3087:3087] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_counters.ini
> [1657513261.979424][3087:3087] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-l8xuPE)
> [1657513261.979534][3087:3087] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
> [1657513261.979726][3087:3087] CHIP:DL: writing settings to file (/tmp/chip_factory.ini-Anil0D)
> [1657513261.980209][3087:3087] CHIP:DL: renamed tmp file to file (/tmp/chip_factory.ini)
> [1657513261.980256][3087:3087] CHIP:DL: NVS set: chip-factory/unique-id = "7F10F63AFCD34573"
> [1657513261.980385][3087:3087] CHIP:DL: writing settings to file (/tmp/chip_factory.ini-FXoI0B)
> [1657513261.980784][3087:3087] CHIP:DL: renamed tmp file to file (/tmp/chip_factory.ini)
> [1657513261.980824][3087:3087] CHIP:DL: NVS set: chip-factory/vendor-id = 65521 (0xFFF1)
> [1657513261.980943][3087:3087] CHIP:DL: writing settings to file (/tmp/chip_factory.ini-jRMkvC)
> [1657513261.981332][3087:3087] CHIP:DL: renamed tmp file to file (/tmp/chip_factory.ini)
> [1657513261.981370][3087:3087] CHIP:DL: NVS set: chip-factory/product-id = 32769 (0x8001)
> [1657513261.981486][3087:3087] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-yupNVA)
> [1657513261.982013][3087:3087] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
> [1657513261.982063][3087:3087] CHIP:DL: NVS set: chip-counters/reboot-count = 1 (0x1)
> [1657513261.982196][3087:3087] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-1KXfJA)
> [1657513261.982611][3087:3087] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
> [1657513261.982654][3087:3087] CHIP:DL: NVS set: chip-counters/total-operational-hours = 0 (0x0)
> [1657513261.982798][3087:3087] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-EOeABD)
> [1657513261.983190][3087:3087] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
> [1657513261.983231][3087:3087] CHIP:DL: NVS set: chip-counters/boot-reason = 0 (0x0)
> [1657513261.983348][3087:3087] CHIP:DL: writing settings to file (/tmp/chip_config.ini-JpTygE)
> [1657513261.983684][3087:3087] CHIP:DL: renamed tmp file to file (/tmp/chip_config.ini)
> [1657513261.983722][3087:3087] CHIP:DL: NVS set: chip-config/regulatory-location = 0 (0x0)
> [1657513261.983841][3087:3087] CHIP:DL: writing settings to file (/tmp/chip_config.ini-dtoKqE)
> [1657513261.984210][3087:3087] CHIP:DL: renamed tmp file to file (/tmp/chip_config.ini)
> [1657513261.984248][3087:3087] CHIP:DL: NVS set: chip-config/location-capability = 2 (0x2)
> [1657513261.984676][3087:3087] CHIP:DL: Failed to get Ethernet interface
> [1657513261.984958][3087:3087] CHIP:DL: Failed to reset Ethernet statistic counts
> [1657513261.985126][3087:3087] CHIP:DL: Got WiFi interface: wlp1s0
> [1657513261.990147][3087:3087] CHIP:DL: Found the primary WiFi interface:wlp1s0
> [1657513261.990253][3087:3087] CHIP:IN: UDP::Init bind&listen port=0
> [1657513261.990327][3087:3087] CHIP:IN: UDP::Init bound to port=50764
> [1657513261.990344][3087:3087] CHIP:IN: UDP::Init bind&listen port=0
> [1657513261.990385][3087:3087] CHIP:IN: UDP::Init bound to port=34728
> [1657513261.990399][3087:3087] CHIP:IN: BLEBase::Init - setting/overriding transport
> [1657513261.990408][3087:3087] CHIP:IN: TransportMgr initialized
> [1657513261.990431][3087:3087] CHIP:FP: Initializing FabricTable from persistent storage
> [1657513261.990471][3087:3087] CHIP:TS: Last Known Good Time: [unknown]
> [1657513261.990488][3087:3087] CHIP:TS: Setting Last Known Good Time to firmware build time 2022-07-07T00:10:25
> [1657513261.994195][3087:3087] CHIP:ZCL: Using ZAP configuration...
> [1657513261.994877][3087:3087] CHIP:CTL: System State Initialized...
> [1657513261.994932][3087:3087] CHIP:CTL: Stopping commissioning discovery over DNS-SD
> [1657513261.994960][3087:3087] CHIP:CTL: Setting attestation nonce to random value
> [1657513261.994993][3087:3087] CHIP:CTL: Setting CSR nonce to random value
> [1657513261.995047][3087:3087] CHIP:IN: UDP::Init bind&listen port=5550
> [1657513261.995092][3087:3087] CHIP:IN: UDP::Init bound to port=5550
> [1657513261.995110][3087:3087] CHIP:IN: UDP::Init bind&listen port=5550
> [1657513261.995146][3087:3087] CHIP:IN: UDP::Init bound to port=5550
> [1657513261.995162][3087:3087] CHIP:IN: TransportMgr initialized
[... cut ...]
> [1657513302.211695][3087:3087] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
> [1657513302.211795][3087:3087] CHIP:DL: NVS set: chip-counters/total-operational-hours = 0 (0x0)
> [1657513302.211816][3087:3087] CHIP:DL: Inet Layer shutdown
> [1657513302.211828][3087:3087] CHIP:DL: BLE shutdown
> [1657513302.213383][3087:3087] CHIP:DL: System Layer shutdown
> [1657513302.213497][3087:3087] CHIP:SPT: VerifyOrDie failure at ../../third_party/connectedhomeip/src/lib/support/Pool.h:317: Allocated() == 0
> Aborted (core dumped)
>

tatsutaigu, 2022-07-11:
@dhrishi 
> Does this happen everytime? 
Yes, it happen everytime.
I will write logs next paragraph.

I tried `all-clusters-app` example.
I do below command for build.

> $ cd $CONNECTEDHOMEIPPATH
> $ source ./scripts/bootstrap.sh
> $ source ./scripts/activate.sh
> $ cd examples/all-clusters-app/esp32/
> $ . ./$ESPIDFPATH/export.sh
> $ mv sdkconfig.defaults sdkconfig_esp32.defaults
> $ ln -s sdkconfig_c3devkit.defaults sdkconfig.defaults
> $ idf.py set-target esp32c3
> $ idf.py build
> $ idf.py -p /dev/ttyUSB0 erase_flash flash

I'm using below branch/commit id for connectedhomeip and ESP-IDF.

* connectedhomeip

> $ git log -1
> commit 126f6b2915fc26ee17342e1dc80f96f031938a6c (HEAD -> master, origin/master, origin/HEAD)
> Author: Stefan Agner <stefan@agner.ch>
> Date:   Wed Jul 6 15:04:20 2022 +0200
>
>    [python] [...]

* ESP-IDF

> $ git log -1
> commit 1329b19fe494500aeb79d19b27cfd99b40c37aec (HEAD, tag: v4.4.1)
> Author: Michael (XIAO Xufeng) <xiaoxufeng@espressif.com>
> Date:   Sat Apr 16 09:43:14 2022 +0800
> 
>     versions: Update version to 4.4.1

tatsutaigu, 2022-07-11:
Logs

* connectedhomeip

> $ ./out/debug/chip-tool pairing ble-wifi 12344321 $SSID $PASSWORD 20202021 3840
> [1657582567.584798][12762:12762] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_kvs
> [1657582567.585477][12762:12762] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_factory.ini
> [1657582567.585790][12762:12762] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_config.ini
> [1657582567.586047][12762:12762] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_counters.ini
> [1657582567.586549][12762:12762] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-wHe3vd)
> [1657582567.587260][12762:12762] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
> [1657582567.587572][12762:12762] CHIP:DL: NVS set: chip-counters/reboot-count = 4 (0x4)
> [1657582567.588278][12762:12762] CHIP:DL: Failed to get Ethernet interface
> [1657582567.588743][12762:12762] CHIP:DL: Failed to reset Ethernet statistic counts
> [1657582567.589151][12762:12762] CHIP:DL: Got WiFi interface: wlp1s0
> [1657582567.590417][12762:12762] CHIP:DL: Found the primary WiFi interface:wlp1s0
> [1657582567.590795][12762:12762] CHIP:IN: UDP::Init bind&listen port=0
> [1657582567.591074][12762:12762] CHIP:IN: UDP::Init bound to port=42583
> [1657582567.591259][12762:12762] CHIP:IN: UDP::Init bind&listen port=0
> [1657582567.591471][12762:12762] CHIP:IN: UDP::Init bound to port=56806
> [1657582567.591651][12762:12762] CHIP:IN: BLEBase::Init - setting/overriding transport
> [1657582567.591873][12762:12762] CHIP:IN: TransportMgr initialized
> [1657582567.592071][12762:12762] CHIP:FP: Initializing FabricTable from persistent storage
> [1657582567.592472][12762:12762] CHIP:TS: Last Known Good Time: 2022-07-07T00:10:25
> [1657582567.593425][12762:12762] CHIP:FP: Fabric index 0x1 was retrieved from storage. Compressed FabricId 0xFDAA879C07F9EFE6, FabricId 0x0000000000000001, NodeId 0x000000000001B669, VendorId 0xFFF1
> [1657582567.594175][12762:12762] CHIP:FP: Fabric index 0x2 was retrieved from storage. Compressed FabricId 0x1E5D8F19A53A5B5E, FabricId 0x0000000000000002, NodeId 0x000000000001B669, VendorId 0xFFF1
> [1657582567.594916][12762:12762] CHIP:FP: Fabric index 0x3 was retrieved from storage. Compressed FabricId 0x087EA5B078BED830, FabricId 0x0000000000000003, NodeId 0x000000000001B669, VendorId 0xFFF1
> [1657582567.599367][12762:12762] CHIP:ZCL: Using ZAP configuration...
> [1657582567.600396][12762:12762] CHIP:CTL: System State Initialized...
> [1657582567.600653][12762:12762] CHIP:CTL: Stopping commissioning discovery over DNS-SD
> [1657582567.600828][12762:12762] CHIP:CTL: Setting attestation nonce to random value
> [1657582567.601021][12762:12762] CHIP:CTL: Setting CSR nonce to random value
> [1657582567.601226][12762:12762] CHIP:IN: UDP::Init bind&listen port=5550
> [1657582567.601424][12762:12762] CHIP:IN: UDP::Init bound to port=5550
> [1657582567.601590][12762:12762] CHIP:IN: UDP::Init bind&listen port=5550
> [1657582567.601776][12762:12762] CHIP:IN: UDP::Init bound to port=5550
> [1657582567.601939][12762:12762] CHIP:IN: TransportMgr initialized
> [1657582567.602366][12762:12762] CHIP:CTL: Stopping commissioning discovery over DNS-SD
> [1657582567.602558][12762:12762] CHIP:CTL: Setting attestation nonce to random value
> [1657582567.602734][12762:12762] CHIP:CTL: Setting CSR nonce to random value
> [1657582567.606064][12762:12762] CHIP:CTL: Generating NOC
> [1657582567.607668][12762:12762] CHIP:FP: Validating NOC chain
> [1657582567.610263][12762:12762] CHIP:FP: NOC chain validation successful
> [1657582567.616436][12762:12762] CHIP:FP: Updated fabric at index: 0x1, Node ID: 0x000000000001B669
> [1657582567.616639][12762:12762] CHIP:TS: Last Known Good Time: 2022-07-07T00:10:25
> [1657582567.616807][12762:12762] CHIP:TS: New proposed Last Known Good Time: 2021-01-01T00:00:00
> [1657582567.616965][12762:12762] CHIP:TS: Retaining current Last Known Good Time
> [1657582567.619844][12762:12762] CHIP:FP: Metadata for Fabric 0x1 persisted to storage.
> [1657582567.622669][12762:12762] CHIP:TS: Committing Last Known Good Time to storage: 2022-07-07T00:10:25
> [1657582567.625302][12762:12762] CHIP:CTL: Joined the fabric at index 1. Compressed fabric ID is: 0x0000000000000000
> [1657582567.625353][12762:12762] CHIP:IN: UDP::Init bind&listen port=5550
> [1657582567.625421][12762:12762] CHIP:IN: UDP::Init bound to port=5550
> [1657582567.625436][12762:12762] CHIP:IN: UDP::Init bind&listen port=5550
> [1657582567.625476][12762:12762] CHIP:IN: UDP::Init bound to port=5550
> [1657582567.625488][12762:12762] CHIP:IN: TransportMgr initialized
> [1657582567.625584][12762:12762] CHIP:CTL: Stopping commissioning discovery over DNS-SD
> [1657582567.625609][12762:12762] CHIP:CTL: Setting attestation nonce to random value
> [1657582567.625645][12762:12762] CHIP:CTL: Setting CSR nonce to random value
> [1657582567.626859][12762:12762] CHIP:CTL: Generating NOC
> [1657582567.628302][12762:12762] CHIP:FP: Validating NOC chain
> [1657582567.630817][12762:12762] CHIP:FP: NOC chain validation successful
> [1657582567.631125][12762:12762] CHIP:FP: Updated fabric at index: 0x2, Node ID: 0x000000000001B669
> [1657582567.631153][12762:12762] CHIP:TS: Last Known Good Time: 2022-07-07T00:10:25
> [1657582567.631165][12762:12762] CHIP:TS: New proposed Last Known Good Time: 2021-01-01T00:00:00
> [1657582567.631174][12762:12762] CHIP:TS: Retaining current Last Known Good Time
[... cut ...]
> I (42863) app-devicecallbacks: Current free heap: 124392
> 
> I (42873) chip[DL]: Configuring CHIPoBLE advertising (interval 500 ms, connectable, device name MATTER-3840)
> I (42883) NimBLE: GAP procedure initiated: advertise; 
> I (42893) NimBLE: disc_mode=2
> I (42893) NimBLE:  adv_channel_map=0 own_addr_type=1 adv_filter_policy=0 adv_itvl_min=800 adv_itvl_max=800
> I (42903) NimBLE: 
> 
> I (42903) chip[DL]: CHIPoBLE advertising started
> I (42913) app-devicecallbacks: Current free heap: 124384

dhrishi, 2022-07-12:
>$ ln -s sdkconfig_c3devkit.defaults sdkconfig.defaults

This is not required. When you do `idf.py set-target esp32c3` the sdkconfig gets updated appropriately. Can you please try with the latest codebase. I am able to get this to work fine with it.

bzbarsky-apple, 2022-07-12:
@tatsutaigu Please attach logs instead of pasting, or at least wrap them in `<details>` bits....

tatsutaigu, 2022-07-12:
@dhrishi 
I tried your suggestion.
However, Timeout occurred.
it may `chip-tool` issue?
for example, `chip-tool` has insufficient control of BLE controller?

@bzbarsky-apple 
I change it.

dhrishi, 2022-07-13:
Ohh, my bad. I am on the latest commit on [my fork](https://github.com/dhrishi/connectedhomeip) which is `https://github.com/dhrishi/connectedhomeip/commit/8ebd7f0fd891dc745b2b2eb9fb5a7ea0b241342b` (around 63 commits behind) and it works there. I will update to the latest codebase and try

dhrishi, 2022-07-13:
@bzbarsky-apple I am able to perform commissioning and cluster control on the current latest commit on upstream connectedhomeip https://github.com/project-chip/connectedhomeip/commit/51e41f9dfb49727a90b9641621085e60f642a884 
Please see attached logs
[esp32c3_51e41f9.txt](https://github.com/project-chip/connectedhomeip/files/9103310/esp32c3_51e41f9.txt)
[chip-tool_51e41f9.txt](https://github.com/project-chip/connectedhomeip/files/9103318/chip-tool_51e41f9.txt)

tatsutaigu, 2022-07-15:
@dhrishi 
Thank for your information.
I tried `51e41f9`, but error occurred.
[chip-tool-51e41f9.txt](https://github.com/project-chip/connectedhomeip/files/9116845/chip-tool-51e41f9.txt)

[esp32-c3-51e41f9.txt](https://github.com/project-chip/connectedhomeip/files/9116846/esp32-c3-51e41f9.txt)

My environment: 
 Ubuntu 20.04
 Celeron N3350 ( [ZBOX PI225 pico](https://www.zotac.com/product/mini_pcs/zbox-pi225-pico-windows-10) )
 This environment successfully completed commissioning with [test_event_4_2021_07_06](https://github.com/project-chip/connectedhomeip/tree/test_event_4_2021_07_06/).

tatsutaigu, 2022-07-15:
@dhrishi

I tried 
`chip-device-ctrl` on `test_event_4_2021_07_06` and `esp32-c3` `all-cluster-app` on `51e41f9`.
[NG_chip-device-ctrl-test_event_4_2021_07_06.txt](https://github.com/project-chip/connectedhomeip/files/9117109/NG_chip-device-ctrl-test_event_4_2021_07_06.txt)
[NG_esp32-c3-51e41f9.txt](https://github.com/project-chip/connectedhomeip/files/9117110/NG_esp32-c3-51e41f9.txt)

`chip-device-ctrl` on `51e41f9` could not run, it aborted.
Therefore, I cannot test `chip-device-ctrl` on `51e41f9`  and `esp32-c3` `all-cluster-app` on `test_event_4_2021_07_06`.

tatsutaigu, 2022-07-19:
@dhrishi 
`all-cluster-app` example on ESP32-C3 is working.
root cause is relation `chip-tool` and bluetooth adapter.

I inserted bluetooth dongle to USB and I choose dongle via `--ble-adapter` option on `chip-tool`, then `Timeout error` does not occur.

current `chip-tool` is degradation from [test_event_4_2021_07_06](https://github.com/project-chip/connectedhomeip/tree/test_event_4_2021_07_06/)
 python `chip-device-ctrl`.

should I change title [ESP32] to [chip-tool]?
and, moderator will change label esp32 to chip-tool?

* hci0 : Timeout Error
[esp32-c3-0718-hci0.txt](https://github.com/project-chip/connectedhomeip/files/9136358/esp32-c3-0718-hci0.txt)
[chiptool-0718-hci0.txt](https://github.com/project-chip/connectedhomeip/files/9136333/chiptool-0718-hci0.txt)

> [1658188151.168122][10375:10380] CHIP:DMG: Time out! failed to receive invoke command response from Exchange: 53514i
> [1658188151.168195][10375:10380] CHIP:CTL: Device failed to receive the CSR request Response: ../../third_party/connectedhomeip/src/app/CommandSender.cpp:214: CHIP Error 0x00000032: Timeout
> [1658188151.168231][10375:10380] CHIP:CTL: Error on commissioning step 'SendOpCertSigningRequest': '../../third_party/connectedhomeip/src/app/CommandSender.cpp:214: CHIP Error 0x00000032: Timeout'
> [1658188151.168246][10375:10380] CHIP:CTL: Failed to perform commissioning step 9
> [1658188151.168260][10375:10380] CHIP:CTL: Going from commissioning step 'SendOpCertSigningRequest' with lastErr = '../../third_party/connectedhomeip/src/app/CommandSender.cpp:214: CHIP Error 0x00000032: Timeout' -> 'Cleanup'
> [1658188151.168285][10375:10380] CHIP:CTL: Performing next commissioning step 'Cleanup' with completion status = '../../third_party/connectedhomeip/src/app/CommandSender.cpp:214: CHIP Error 0x00000032: Timeout'

* hic1 : No Error
[esp32-c3-0718-hci1.txt](https://github.com/project-chip/connectedhomeip/files/9136359/esp32-c3-0718-hci1.txt)
[chiptool-0718-hci1.txt](https://github.com/project-chip/connectedhomeip/files/9136329/chiptool-0718-hci1.txt)

* connectedhomeip

> $ git log -1
> commit aa9457e6b94b735076dff6297176183bf9780177 (HEAD -> master, origin/sve, origin/master, origin/HEAD)
> Author: manjunath-grl <102359958+manjunath-grl@users.noreply.github.com>
> Date:   Sun Jul 17 07:47:19 2022 +0530
> 
>     Manual and modified script update (#20821)

* bluetooth interface on PC

> $ hciconfig --all
> hci1:	Type: Primary  Bus: USB
> 	BD Address: 00:1B:DC:FE:D1:61  ACL MTU: 310:10  SCO MTU: 64:8
> 	UP RUNNING 
> 	RX bytes:6232 acl:171 sco:0 events:135 errors:0
> 	TX bytes:6200 acl:33 sco:0 commands:90 errors:0
> 	Features: 0xff 0xff 0x8f 0xfe 0xdb 0xff 0x5b 0x87
> 	Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3 
> 	Link policy: RSWITCH HOLD SNIFF PARK 
> 	Link mode: SLAVE ACCEPT 
> 	Name: 'tatsutaigu-ZBOX-PI225'
> 	Class: 0x0c0104
> 	Service Classes: Rendering, Capturing
> 	Device Class: Computer, Desktop workstation
> 	HCI Version: 4.0 (0x6)  Revision: 0x22bb
> 	LMP Version: 4.0 (0x6)  Subversion: 0x22bb
> 	Manufacturer: Cambridge Silicon Radio (10)
> 
> hci0:	Type: Primary  Bus: USB
> 	BD Address: F4:96:34:8F:7E:96  ACL MTU: 1021:5  SCO MTU: 96:6
> 	UP RUNNING 
> 	RX bytes:2330 acl:0 sco:0 events:253 errors:0
> 	TX bytes:41862 acl:0 sco:0 commands:252 errors:0
> 	Features: 0xff 0xfe 0x0f 0xfe 0xdb 0xff 0x7b 0x87
> 	Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3 
> 	Link policy: RSWITCH HOLD SNIFF 
> 	Link mode: SLAVE ACCEPT 
> 	Name: 'tatsutaigu-ZBOX-PI225 #1'
> 	Class: 0x0c0104
> 	Service Classes: Rendering, Capturing
> 	Device Class: Computer, Desktop workstation
> 	HCI Version: 4.2 (0x8)  Revision: 0x1000
> 	LMP Version: 4.2 (0x8)  Subversion: 0x1000
> 	Manufacturer: Intel Corp. (2)
>

tatsutaigu, 2022-07-21:
Hi,
I use below commit ID, then It is resolved.

commit 513235823b2a1f6b5a299f71feef9e28c95dc1cf (HEAD -> master, origin/master, origin/HEAD)

closed 2022-07-21 by tatsutaigu