project-chip/connectedhomeip #20481: CHIP: [BLE] Service not found on the device

JourneyYoung, 2022-07-08:
#### Problem
<what's wrong or missing, please include any applicable:
 * expected behavior
 * actual behavior
 * steps to reproduce
 * system configuration
 * screenshots, images, logs, etc.
 >

An iphone12 running on iOS 15.3.1 can not find FFF6 Service. I used two iphone12 to connect the same matter device,  the other one was running on iOS 15.4. When I connected to the matter ble device, the one running on iOS 15.4 can successed create PASE Session, but the other iphone running on 15.3.1 cannot find FFF6 service, then failed to create PASE Session, the error info is "CHIP: [BLE] Service not found on the device" . So I used nRF connect app to look up Matter blutooth service, see as these pictures:

[image]

[image]

The failed one can not find FFF6 service. Is this an iOS's bug？How can I fix it?

 #### Proposed Solution
<suggested fix, suggested enhancement>

andy31415, 2022-07-12:
marked darwin to tag iOS developers. @woody-apple - we do not seem to have a iOS tag (unsure if needed or not)

JourneyYoung, 2022-07-12:
您好。你的邮件我已经收到，谢谢！

woody-apple, 2022-07-19:
@JourneyYoung thank you for the report! Can you please let us know what version of the framework the device is using? Does this happen on 15.6 developer releases?

closed 2022-07-19 by woody-apple