project-chip/connectedhomeip #20496: Improve REPL test useability

mrjerryjohns, 2022-07-08:
#### Problem

The REPL tests have come a long way in terms of usability, but it still has a number of key improvements remaining to make them truly usable:

1. Better isolation of each test in terms of setting up pre-requisites. Right now, each test depends on an implicit running of a previous test to setup the target with a number of fabrics. It'd be great if a test could express a pre-requisite for the target state and have the test framework isolate the tracking of the target's state.
2. Ability to just run a specific test.
3. Ability to quickly load up a test in the REPL itself and run it against a externally launched server application.

mrjerryjohns, 2022-07-08:
4. There are negative tests we'd like to add sometimes that take a long time to actually fail correctly (timeouts for example), that we don't want to actually run when in 'developer' mode, but do want to run in CI. Would be great if we could add a global option that each test can access to figure out whether it should run such a test.

turon, 2022-07-20:
Perhaps make this a task list by appending `- [ ] ` to each item and consolidating the four items in the description?

closed 2023-02-05 by stale[bot]