project-chip/connectedhomeip #20537: [ESP32C3] cannot finish commissioning

tatsutaigu, 2022-07-11:
#### Problem
 * expected behavior
 * actual behavior
 I set up and follow sequence about [all-clusters-app](https://github.com/project-chip/connectedhomeip/tree/master/examples/all-clusters-app/esp32)
 Run `chip-tool` but chip-tool was stopped by timeout.
 How do I complete commissioning?
 If I need to modify menuconfig, please let me know setting on menuconfig.

 * steps to reproduce
 * system configuration

example : `all-clusters-app`.
ESP32 : ESP32-C3 (ESP32-C3-DevkitM-1)

connectedhomeip : master
> $ git log -1
> commit 776c06d5bbda9ea59f8eb28a8d328251fe619d61 (HEAD -> master, origin/master, origin/HEAD)
> Author: Shubham Patil <shubham.patil@espressif.com>
> Date:   Sat Jul 9 18:23:18 2022 +0530
> 
>     [ESP32] Support flash encryption and nvs encryption (#20414)
>     
>     - Added an option to encrypt the factory partition.
>     - Securely initialize the NVS partition if NVS encryption is enabled
>     - Added user guide for flash encryption in lighting-app/esp32

ESP-IDF : v4.4.1

> $ git log -1
> commit 1329b19fe494500aeb79d19b27cfd99b40c37aec (HEAD, tag: v4.4.1)
> Author: Michael (XIAO Xufeng) <xiaoxufeng@espressif.com>
> Date:   Sat Apr 16 09:43:14 2022 +0800
> 
>     versions: Update version to 4.4.1

 * screenshots, images, logs, etc.

* chip-tool log

> $ ./out/debug/chip-tool pairing ble-wifi 12344321 $SSID $PASSWORD 20202021 3840
> [1657520561.895036][21740:21740] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_kvs
> [1657520561.895478][21740:21740] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_factory.ini
> [1657520561.895627][21740:21740] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_config.ini
> [1657520561.895737][21740:21740] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_counters.ini
> [1657520561.896225][21740:21740] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-se6Q34)
> [1657520561.896933][21740:21740] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
> [1657520561.897038][21740:21740] CHIP:DL: NVS set: chip-counters/reboot-count = 7 (0x7)
> [1657520561.897595][21740:21740] CHIP:DL: Failed to get Ethernet interface
> [1657520561.897961][21740:21740] CHIP:DL: Failed to reset Ethernet statistic counts
> [1657520561.898143][21740:21740] CHIP:DL: Got WiFi interface: wlp1s0
> [1657520561.899441][21740:21740] CHIP:DL: Found the primary WiFi interface:wlp1s0
> [1657520561.899574][21740:21740] CHIP:IN: UDP::Init bind&listen port=0
> [1657520561.899693][21740:21740] CHIP:IN: UDP::Init bound to port=52158
> [1657520561.899739][21740:21740] CHIP:IN: UDP::Init bind&listen port=0
> [1657520561.899823][21740:21740] CHIP:IN: UDP::Init bound to port=56227
> [1657520561.899868][21740:21740] CHIP:IN: BLEBase::Init - setting/overriding transport
> [1657520561.899909][21740:21740] CHIP:IN: TransportMgr initialized
> [1657520561.899966][21740:21740] CHIP:FP: Initializing FabricTable from persistent storage
> [1657520561.900239][21740:21740] CHIP:TS: Last Known Good Time: 2022-07-11T15:15:36
> [1657520561.901082][21740:21740] CHIP:FP: Fabric index 0x1 was retrieved from storage. Compressed FabricId 0xC514CBA34B4E423E, FabricId 0x0000000000000001, NodeId 0x000000000001B669, VendorId 0xFFF1
> [1657520561.901678][21740:21740] CHIP:FP: Fabric index 0x2 was retrieved from storage. Compressed FabricId 0xE9F8A4A0812BB2F3, FabricId 0x0000000000000002, NodeId 0x000000000001B669, VendorId 0xFFF1
> [1657520561.902353][21740:21740] CHIP:FP: Fabric index 0x3 was retrieved from storage. Compressed FabricId 0x8EB7D3E2000C9D1D, FabricId 0x0000000000000003, NodeId 0x000000000001B669, VendorId 0xFFF1
> [1657520561.905426][21740:21740] CHIP:ZCL: Using ZAP configuration...
> [1657520561.906207][21740:21740] CHIP:CTL: System State Initialized...
> [1657520561.906304][21740:21740] CHIP:CTL: Stopping commissioning discovery over DNS-SD
> [1657520561.906352][21740:21740] CHIP:CTL: Setting attestation nonce to random value
> [1657520561.906410][21740:21740] CHIP:CTL: Setting CSR nonce to random value
> [1657520561.906487][21740:21740] CHIP:IN: UDP::Init bind&listen port=5550
> [1657520561.906556][21740:21740] CHIP:IN: UDP::Init bound to port=5550
> [1657520561.906597][21740:21740] CHIP:IN: UDP::Init bind&listen port=5550
[... 403 lines cut ...]
> I (2232) NimBLE: GAP procedure initiated: stop advertising.
> 
> I (2242) NimBLE: GAP procedure initiated: advertise; 
> I (2242) NimBLE: disc_mode=2
> I (2252) NimBLE:  adv_channel_map=0 own_addr_type=1 adv_filter_policy=0 adv_itvl_min=40 adv_itvl_max=40
> I (2262) NimBLE: 
> 
> I (2262) chip[SVR]: Cannot load binding table: a0
> I (8232) chip[DL]: BLE GAP connection established (con 1)
> I (8232) chip[DL]: CHIPoBLE advertising stopped
> I (8232) app-devicecallbacks: Current free heap: 124384
> 
> I (40442) chip[DL]: BLE GAP connection terminated (con 1 reason 0x213)
> I (40452) app-devicecallbacks: CHIPoBLE disconnected
> I (40452) app-devicecallbacks: Current free heap: 124384
> 
> I (40452) chip[DL]: Configuring CHIPoBLE advertising (interval 500 ms, connectable, device name MATTER-3840)
> I (40462) NimBLE: GAP procedure initiated: advertise; 
> I (40472) NimBLE: disc_mode=2
> I (40472) NimBLE:  adv_channel_map=0 own_addr_type=1 adv_filter_policy=0 adv_itvl_min=800 adv_itvl_max=800
> I (40482) NimBLE: 
> 
> I (40492) chip[DL]: CHIPoBLE advertising started
> I (40492) app-devicecallbacks: Current free heap: 124376
>

sinny777, 2022-07-13:
I'm also facing the same problem....

tatsutaigu, 2022-07-19:
I solved this issue.

https://github.com/project-chip/connectedhomeip/issues/20470#issuecomment-1181443365

https://github.com/project-chip/connectedhomeip/issues/20470#issuecomment-1181443365

closed 2022-07-19 by tatsutaigu