project-chip/connectedhomeip #20580: Compilation error on M1 Mac while trying to build darwin-framework-tool

phonnakasturi-apple, 2022-07-11:
With SHA caf62f709e0a0fdf41b0d3a1652cc26bc46a960a

While trying to build _darwin-framework-tool_ on M1 Mac, we encounter the following complier error;

**Steps to reproduce**

1. cd ~/connectedhomeip
2. git reset --hard caf62f709e0a0fdf41b0d3a1652cc26bc46a960a
3. export PKG_CONFIG_PATH="/opt/homebrew/opt/openssl@3/lib/pkgconfig"
4. source scripts/activate.sh
5. cd examples/darwin-framework-tool
6. gn gen out/debug --args='chip_mdns="platform" chip_crypto="mbedtls" chip_inet_config_enable_ipv4=false'
7. ninja -C out/debug

```
**Error message**
FAILED: macos_framework_output/Matter.framework 
python ../../third_party/connectedhomeip/scripts/build/build_darwin_framework.py --project_path ../../third_party/connectedhomeip/src/darwin/Framework/Matter.xcodeproj --out_path macos_framework_output --target Matter --log_path darwin_framework_build.log
2022-07-08 17:19:38.507 xcodebuild[17102:4597155] Requested but did not find extension point with identifier Xcode.IDEKit.ExtensionSentinelHostApplications for extension Xcode.DebuggerFoundation.AppExtensionHosts.watchOS of plug-in com.apple.dt.IDEWatchSupportCore
2022-07-08 17:19:38.511 xcodebuild[17102:4597155] Requested but did not find extension point with identifier Xcode.IDEKit.ExtensionPointIdentifierToBundleIdentifier for extension Xcode.DebuggerFoundation.AppExtensionToBundleIdentifierMap.watchOS of plug-in com.apple.dt.IDEWatchSupportCore
2022-07-08 17:19:38.723 xcodebuild[17102:4597155] DVTCoreDeviceEnabledState: DVTCoreDeviceEnabledState_Disabled set via user default (DVTEnableCoreDevice=(null))
** BUILD FAILED **

The following build commands failed:
	PhaseScriptExecution Run\ Script /Users/priyankahonnakasturi/connectedhomeip/src/darwin/Framework/build/Matter.build/Release/Matter.build/Script-0C40A67D246C9AC700885C81.sh (in target 'Matter' from project 'Matter')
(1 failure)
Running xcodebuild -target Matter -sdk macosx -project ../../third_party/connectedhomeip/src/darwin/Framework/Matter.xcodeproj CONFIGURATION_BUILD_DIR=/Users/priyankahonnakasturi/connectedhomeip/examples/darwin-framework-tool/out/debug/macos_framework_output
Traceback (most recent call last):
 File "/Users/priyankahonnakasturi/connectedhomeip/examples/darwin-framework-tool/out/debug/../../third_party/connectedhomeip/scripts/build/build_darwin_framework.py", line 63, in <module>
  build_darwin_framework(args)
 File "/Users/priyankahonnakasturi/connectedhomeip/examples/darwin-framework-tool/out/debug/../../third_party/connectedhomeip/scripts/build/build_darwin_framework.py", line 32, in build_darwin_framework
  command_result = run_command(command)
 File "/Users/priyankahonnakasturi/connectedhomeip/examples/darwin-framework-tool/out/debug/../../third_party/connectedhomeip/scripts/build/build_darwin_framework.py", line 22, in run_command
  return str(subprocess.check_output(command.split()))
 File "/opt/homebrew/Cellar/python@3.9/3.9.13_1/Frameworks/Python.framework/Versions/3.9/lib/python3.9/subprocess.py", line 424, in check_output
  return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
 File "/opt/homebrew/Cellar/python@3.9/3.9.13_1/Frameworks/Python.framework/Versions/3.9/lib/python3.9/subprocess.py", line 528, in run
  raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['xcodebuild', '-target', 'Matter', '-sdk', 'macosx', '-project', '../../third_party/connectedhomeip/src/darwin/Framework/Matter.xcodeproj', 'CONFIGURATION_BUILD_DIR=/Users/priyankahonnakasturi/connectedhomeip/examples/darwin-framework-tool/out/debug/macos_framework_output']' returned non-zero exit status 65.
[85/609] c++ obj/third_party/connected...xamples/chip-tool/chip-tool.main.cpp.o

```

By editing the file chip_xcode_build_connector and manually adding;
`PLATFORM_PREFERRED_ARCH=arm64`
resolved this issue

krypton36, 2022-07-11:
It appears that this [PR](https://github.com/project-chip/connectedhomeip/pull/9353/files) may be the culprit.

woody-apple, 2022-07-19:
May be resolved with: https://github.com/project-chip/connectedhomeip/pull/20952

phonnakasturi-apple, 2022-07-19:
@woody-apple Just confirmed with SHA 56db839c1dd7c8426a060cbc0c9d68dc4145dbf5
The issue is still present

phonnakasturi-apple, 2022-08-04:
This issue is still present

PR #21761 (2022-08-09): Resolve compilation issue for darwin-framework-tool on Apple Silicon
#### Problem
* Fix compilation issue for Darwin Framework tool on Apple Silicon
* Fixes #20580 

#### Change overview
- Updates python script to set PLATFORM_PREFERRED_ARCH

#### Testing
Compiled on Intel and Apple Silicon
closed 2022-08-09 by woody-apple