project-chip/connectedhomeip #20591: [OTA Matter.framework] Code is unsafe/racy error message

krypton36, 2022-07-12:
#### Problem
After setting the OTA delegate in Matter.framework and receiving Query Image InvokeCommand to the controller (Matter.framework), an error message states that the code is unsafe/racy. This appears to be the issue because during the transmission of the message we are using a delegate that is passed in from the caller [here](https://github.com/project-chip/connectedhomeip/pull/20516/files#diff-13bd030c2b38e3e7d4738070dd3c919bbdfe9ec8b39cc29996989936150e6d1fR59).  After inspecting the code, the assertion calls IsChipStackLockedByCurrentThread to check whether messaging is locked. For this to be set to true the following condition needs to be true `!mWorkQueue || mIsWorkQueueSuspended || dispatch_get_current_queue() == mWorkQueue` 

 #### Proposed Solution
Update the queue used to send the message to the node. 

[controller_error_log.txt](https://github.com/project-chip/connectedhomeip/files/9089424/controller_error_log.txt)

bzbarsky-apple, 2022-07-12:
The `completionHandler` there should be dispatching to the Matter work queue, no?  @carol-apple

krypton36, 2022-07-12:
@carol-apple I am going to update this.

PR #20628 (2022-07-13): Update OTA queues to use the matter work queue.
#### Problem
There is a racy/unsafe scenario when using OTA software update in the Matter.framework. 
- Fixes #20591 

#### Change overview
- Updates the dispatch queue to use the matter work queue.

#### Testing
- Ran scenario that causes this issue and no longer reproduced.
PR #20690 (2022-07-13): Update OTA queues to use the matter work queue.
#### Problem
There is a racy/unsafe scenario when using OTA software update in the Matter.framework. 
- Fixes #20591 

#### Change overview
- Updates the dispatch queue to use the matter work queue.

#### Testing
- Ran scenario that causes this issue and no longer reproduced.
closed 2022-07-13 by andy31415