project-chip/connectedhomeip #20607: [SVE1] zap sets old cluster revisions by default

markus-becker-tridonic-com, 2022-07-12:
#### Problem

When creating a new EP and e.g. create an Identify Server Cluster, zap fills in version 2 for the Cluster Revision of Identify. Current value is 4.

 #### Proposed Solution

Set the latest ClusterRevision by default.

markus-becker-tridonic-com, 2022-07-12:
I noticed this only for Identify when adding a new endpoint and adding the server cluster. Might not be limited to Identify Server cluster though. Yes, I was using the ZAP UI.

andy31415, 2022-07-12:
Maybe this should be reported to zap? @tecimovic - how do we triage and split bugs between zap and matter sdk?

andy31415, 2022-07-12:
Assigning myself to validate that our XML is correct. Assume that zap takes the value from the xml.

andy31415, 2022-07-12:
illuminance-measurement-cluster.xml has 2 instead of 3

andy31415, 2022-07-12:
preassure-measurement has 2 instead of 3

andy31415, 2022-07-12:
Created a diffs script, revisions mismatches:
0x0004 should be 4 not 3
0x0301 should be 4 not 3
0x0406 should be 3 not 2

Several other clusters I could not properly parse spec vs code. The spec adocs are  too free form.

PR #20636 (2022-07-13): Update cluster default revision based on spec
#### Problem
Spec has updated cluster revision numbers. Match those numbers in the xml.

Fixes #20607

#### Change overview
Updated a few cluster default values for revision numbers.

#### Testing
N/A - compared with spec.
closed 2022-07-13 by andy31415