project-chip/connectedhomeip #20614: [SVE1] test script LVL3.1 needs change in PICS condition

leorozendaal, 2022-07-12:
#### Problem
context: https://csamembers.slack.com/archives/C03NYBWMT42/p1657629815867219

(Jan Claes): LVL3.1 Test will try to read the current level in step 16 while the PICS says the OnOffTransistion is not supported and it did not set the level in Test Step 14 : sends a Move to level command as this is skipped due to the PICS

after comparison of test script https://github.com/project-chip/connectedhomeip/blob/master/src/app/tests/suites/certification/Test_TC_LVL_3_1.yaml to the test plan, it turns out the 16th step (line 174) needs this PICS condition: `LVL.S.C00.Rsp & LVL.S.C00.Rsp && LVL.S.A0010 `

 #### Proposed Solution
change script accordingly

raju-apple, 2022-07-12:
@kvikrambhat Can you add this to the list as well

manjunath-grl, 2022-07-14:
@leorozendaal fixed in PR [20709](https://github.com/project-chip/connectedhomeip/pull/20709) please verify it and close the ticket

closed 2022-07-19 by woody-apple