project-chip/connectedhomeip #20626: [crash] CHIP:  chip::System::TimerPool<chip::System::TimerList::Node>::Invoke

aajain-com, 2022-07-12:
#### Problem

Found the following crash during stability testing with SDK SHA `5d8599d195a93dfe5473f64e4f888c322e5df1d2 `

```
Thread 4 Crashed [↩](https://crashtracer.ios.apple.com/report/show?app=18343&build=20J336&signature=73686696&users=all#log_sections)::   Dispatch queue: com.zigbee.chip.framework.controller.workqueue
0                                 	       0x140e90003
1   CHIP                          	       0x1c62c0834        chip::System::TimerPool<chip::System::TimerList::Node>::Invoke(chip::System::TimerList::Node*) + 88 (Sources/CHIPFramework/connectedhomeip/src/system/SystemTimer.h:224)
2   CHIP                          	       0x1c62bf488        chip::System::LayerImplSelect::HandleTimerComplete(chip::System::TimerList::Node*) + 60 (Sources/CHIPFramework/connectedhomeip/src/system/SystemLayerImplSelect.cpp:421)
3   CHIP                          	       0x1c62bf440        ___ZN4chip6System15LayerImplSelect10StartTimerENSt3__16chrono8durationIjNS2_5ratioILl1ELl1000EEEEEPFvPNS0_5LayerEPvESA__block_invoke + 80 (Sources/CHIPFramework/connectedhomeip/src/system/SystemLayerImplSelect.cpp:148)
```

bzbarsky-apple, 2022-07-14:
```
3   CHIP                          	       0x1c62bf440        ___ZN4chip6System15LayerImplSelect10StartTimerENSt3__16chrono8durationIjNS2_5ratioILl1ELl1000EEEEEPFvPNS0_5LayerEPvESA__block_invoke + 80 (Sources/CHIPFramework/connectedhomeip/src/system/SystemLayerImplSelect.cpp:148)
```
is inside `LayerImplSelect::StartTimer` so this is not the issue that https://github.com/project-chip/connectedhomeip/pull/20704 was fixing...

turon, 2022-07-20:
What are the replication steps?

bzbarsky-apple, 2022-07-20:
@turon This is coming from an automated crash reporting system, so the best we might have for steps so far is "Run a Matter device for a while".

woody-apple, 2022-08-30:
This is no longer showing up in crashes.

closed 2022-08-30 by woody-apple