project-chip/connectedhomeip #20656: ble-wifi pairing failing BlueZ disconnecting

chrisbeach59, 2022-07-12:
BlueZ disconnecting during pairing with the SVE1.1 image.  This was working fine with SVE1.0

Log from pairing using the chip-tool attached.
[pairing.log](https://github.com/project-chip/connectedhomeip/files/9096864/pairing.log)

mykrupp, 2022-07-13:
Silabs running into an issue today. Pairing with Matter over wifi. We are also experiencing issues with SVE1.1, while we had success with preliminary tests running the 1.0 image.

chip-tool pairing logs, running TC DRLK 2.2
[UI_logs (3).log](https://github.com/project-chip/connectedhomeip/files/9097694/UI_logs.3.log)

princemasu, 2022-10-19:
We also observed BlueZ disconnecting during pairing with Matter over WiFi. We used raspberry pi as Matter accessory.
[pairing.log](https://github.com/project-chip/connectedhomeip/files/9816226/pairing.log)

jmartinez-silabs, 2022-11-01:
Something I noticed personally is that bluez is really flaky on raspis when connected to a 2.4ghz wifi. more so with 22.04 Ubuntu than previous version. I had a lot more success when my Pi was connected through ethernet or a 5ghz AP

closed 2022-11-17 by woody-apple