project-chip/connectedhomeip #20668: [Android] How to get ChipDeviceController node id?

Allen-Guof, 2022-07-13:
I want to upgrade firmware through Android phone as ota provider. The firmware upgrade package is in the local file. But when announceOtaProvider, I can't find providerNodeId. I'm guessing that providerNodeId is the ChipDeviceController node id. Is there anyone who can help me.

TheWalkingDead1024, 2022-08-02:
@vidhis88 @mspang michael @Damian-Nordic @markus-becker-tridonic-com @austinh0 @ritikananda27 @wgtdkp @vivien-apple @chulspro @pan-apple @g-coppock @joonhaengHeo @woody-apple @emargolis @joonhaengHeo @gmarcosb @chrisdecenzo, could anybody have a look at this issue？Many thanks!

TheWalkingDead1024, 2022-08-03:
https://github.com/project-chip/connectedhomeip/issues/21517

closed 2023-02-12 by stale[bot]

Allen-Guof, 2023-08-17:
@yunhanw-google Hi, is there any update on this issue. If the issue is not resolved, please help me change the status of the issue. I do not have the permission.

TheWalkingDead1024, 2023-08-18:
@https://github.com/yunhanw-google Could you please take a look at this GitHub issue?

TheWalkingDead1024, 2023-08-18:
@yunhanw-google Could you please take a look at this GitHub issue?