project-chip/connectedhomeip #20671: [TC-ULABEL-2.3]--When TH writes LabelList attribute of constraint more the 16 bytes, DUT responds as Failure instead CONSTRAINT_ERROR

Rajashreekalmane, 2022-07-13:
#### Problem
As per test-plan if TH write  LabelList attribute of constraint more the 16 bytes DUT Should responds with  status code as CONSTRAINT_ERROR but currently DUT is responding as Failure.

 * expected behavior:-CONSTRAINT_ERROR(0x87)
 * actual behavior:-FAILURE(0x01)
 * steps to reproduce:-./chip-tool userlabel write label-list '[{"label":"roomName", "value":"master bedroom 1"}, {"label":"orientationorientation", "value":"east"}, {"label":"floor", "value":"2"}, {"label":"roomType", "value":"bedroom"}]' 1 0

system configuration
Chip-tool - RPI-4, 8GB RAM
DUT - RPI - RPI-4, 8GB RAM
Platform : IP and wifi
Commit-ID:dad96b1e46c98b3446af9b4bfe22593fdfb02b22
Dut application: all-clusters-app

**Test-Plan reference:-**
https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/data_model/Label-Cluster.adoc#data-types

**Spec Reference:-**
https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/data_model/Label-Cluster.adoc#user-label-cluster

**Spec screenshot:-**
[image]

andy31415, 2022-07-13:
This has been fixed in master yesterday.

andy31415, 2022-07-13:
Fix was in #20553

andy31415, 2022-07-13:
I believe it has to be manually cherrypicked - it had no SVE tag :(

andy31415, 2022-07-13:
https://github.com/project-chip/connectedhomeip/pull/20680 for a manual cherrypick

andy31415, 2022-07-13:
Reopened to mark fixed only once CP lands in sve branch.

PR #20680 (2022-07-13): Cherrypick: Return CONSTRAINT_ERROR on user labels that are too long (instead of …
…FAILURE) (#20553)

* Mark constraint error as the error to return on overly long values

* Zap regen

* Fix return code

* Update TC-ULABEL-2.3 for using constraint_error

* Zap regen

This CP is to make #20671 pass in sve branch.
Fixes #20671