project-chip/connectedhomeip #20677: No implementation of Pulse Width Modulation cluster

bzbarsky-apple, 2022-07-13:
#### Problem
There is no implementation of the Pulse Width Modulation cluster, even though it's defined in the appclusters spec.

 #### Proposed Solution
Either implement this, or remove it from the 1.0 spec.

@anush-apple @andy31415 @woody-apple @robszewczyk

bzbarsky-apple, 2022-07-13:
Also no test plan for this; see https://github.com/CHIP-Specifications/chip-test-plans/issues/1736

andy31415, 2022-07-13:
I think we need to improve our heavy overhead of implementing new clusters. This may be post 1.0, but implementing a cluster which is basically read/write some attributes and handle some commands should be trivial. I don't think it is trivial enough yet (it sounds like it: have an xml and add it to something like all-clusters ... I think more is involved).

bzbarsky-apple, 2022-07-13:
I suspect this will end up getting marked provisional in the spec in practice....

woody-apple, 2022-07-20:
Spec Review: We should follow up with a PR in the spec to mark this provisional. @bzbarsky-apple can you file a spec issue to cover say this needs to be marked provisional?

closed 2022-07-20 by bzbarsky-apple