project-chip/connectedhomeip #20678: Mac all cluster app crashes when tried to pair a second fabric (test app) after first adding in Home app

vinay-apple, 2022-07-13:
#### Problem
Mac all cluster app crashes when tried to pair a second fabric (test app) after first adding in Home app 

SHA: 5d8599d195a93dfe5473f64e4f888c322e5df1d2

Steps:

1. Pair Mac all cluster app in Home app
2. Open pairing and copy the setup payload
3. Try pairing with test app Chai and see that Mac all cluster crashes. 

Seeing no memory error before the crash. 

 022-07-12 22:49:02.522291-0700 chip-all-clusters-app[15985:8133943] [all
 ] 🔴 [1657691342522] [15985:8133943] CHIP: [DMG] Error retrieving data
 from clusterId: 0x0000_001D, err = ../../examples/all-clusters-app/linux
 /third_party/connectedhomeip/src/system/TLVPacketBufferBackingStore.cpp:
 88: CHIP Error 0x0000000B: No memory
 [1657691342522] [15985:8133943] CHIP: [DMG] Error retrieving data from
 clusterId: 0x0000_001D, err = ../../examples/all-clusters-app/linux/third
 _party/connectedhomeip/src/system/TLVPacketBufferBackingStore.cpp:88: CHIP
 Error 0x0000000B: No memory
 2022-07-12 22:49:02.522301-0700 chip-all-clusters-app[15985:8133943] [all
 ] 🟢 [1657691342522] [15985:8133943] CHIP: [DMG] <RE:Run> We cannot put
 more chunks into this report. Enable chunking.
 [1657691342522] [15985:8133943] CHIP: [DMG] <RE:Run> We cannot put more
 chunks into this report. Enable chunking.
 
Further down seeing crash with reason noted. 

 2022-07-12 22:50:39.957514-0700 chip-all-clusters-app[15985:8140025] [all
] 🟢 [1657691439957] [15985:8140025] CHIP: [IN] SecureSession[0x10028a6f0
 ]: Released - Type:2 LSID:4709
 [1657691439957] [15985:8140025] CHIP: [IN] SecureSession[0x10028a6f0]:
 Released - Type:2 LSID:4709
 Process 15985 stopped
 * thread #4, queue = 'com.zigbee.chip.framework.controller.workqueue',
 stop reason = EXC_BAD_ACCESS (code=2, address=0x100130003)
     frame #0: 0x0000000100130003 chip-all-clusters-app`chip::app::WriteHandler:
 :ProcessWriteRequest(this=0x0000700003424868, aPayload=0x0000000100c0b730,
 aIsTimedWrite=false) at WriteHandler.cpp:621:5
    618 	    {
    619 	        err = ProcessAttributeDataIBs(AttributeDataIBsReader);
    620 	    }
 -> 621 	    SuccessOrExit(err);
    622 	    SuccessOrExit(err = writeRequestParser.ExitContainer());
    623 	
    624 	    if (err == CHIP_NO_ERROR)
 Target 0: (chip-all-clusters-app) stopped.
 (lldb)

Attached detailed logs for the test. 

[mac-all-cluster-lldb-logs-crash-after-chai-pairing.txt](https://github.com/project-chip/connectedhomeip/files/9103220/mac-all-cluster-lldb-logs-crash-after-chai-pairing.txt)

bzbarsky-apple, 2022-07-13:
@vinay-apple ran this with asan turned on and got:

<details>
<summary>`==40999==ERROR: AddressSanitizer: heap-use-after-free on address 0x60400012c678 at pc 0x0001003dff2a bp 0x700006c98f90 sp 0x700006c98f88`</summary>

```
WRITE of size 8 at 0x60400012c678 thread T7
    #0 0x1003dff29 in chip::System::TimerList::Remove(chip::System::TimerList::Node*) SystemTimer.cpp:92
    #1 0x1003e4e37 in chip::System::LayerImplSelect::HandleTimerComplete(chip::System::TimerList::Node*) SystemLayerImplSelect.cpp:420
    #2 0x1003e58ef in invocation function for block in chip::System::LayerImplSelect::ScheduleWork(void (*)(chip::System::Layer*, void*), void*) SystemLayerImplSelect.cpp:197
    #3 0x1012b81ba in __wrap_dispatch_async_block_invoke+0xca (libclang_rt.asan_osx_dynamic.dylib:x86_64h+0x4a1ba)
    #4 0x7fff202b9622 in _dispatch_call_block_and_release+0xb (libdispatch.dylib:x86_64+0x2622)
    #5 0x7fff202ba805 in _dispatch_client_callout+0x7 (libdispatch.dylib:x86_64+0x3805)
    #6 0x7fff202c05e9 in _dispatch_lane_serial_drain+0x25d (libdispatch.dylib:x86_64+0x95e9)
    #7 0x7fff202c10ac in _dispatch_lane_invoke+0x16d (libdispatch.dylib:x86_64+0xa0ac)
    #8 0x7fff202cac0c in _dispatch_workloop_worker_thread+0x32a (libdispatch.dylib:x86_64+0x13c0c)
    #9 0x7fff2046145c in _pthread_wqthread+0x139 (libsystem_pthread.dylib:x86_64+0x345c)
    #10 0x7fff2046042e in start_wqthread+0xe (libsystem_pthread.dylib:x86_64+0x242e)

0x60400012c678 is located 40 bytes inside of 48-byte region [0x60400012c650,0x60400012c680)
freed by thread T7 here:
    #0 0x1012b90c9 in wrap_free+0xa9 (libclang_rt.asan_osx_dynamic.dylib:x86_64h+0x4b0c9)
    #1 0x1003da446 in chip::Platform::MemoryFree(void*) CHIPMem-Malloc.cpp:111
    #2 0x1003f45b4 in void chip::Platform::Delete<chip::System::TimerList::Node>(chip::System::TimerList::Node*) CHIPMem.h:164
    #3 0x1003f4533 in chip::HeapObjectPool<chip::System::TimerList::Node>::ReleaseObject(chip::System::TimerList::Node*) Pool.h:359
    #4 0x1003e50a8 in chip::System::TimerPool<chip::System::TimerList::Node>::Release(chip::System::TimerList::Node*) SystemTimer.h:205
    #5 0x1003e4fa7 in chip::System::LayerImplSelect::CancelTimer(void (*)(chip::System::Layer*, void*), void*) SystemLayerImplSelect.cpp:178
    #6 0x1003e52d3 in chip::System::LayerImplSelect::ScheduleWork(void (*)(chip::System::Layer*, void*), void*) SystemLayerImplSelect.cpp:186
    #7 0x10038ae87 in chip::app::reporting::Engine::ScheduleRun() Engine.cpp:567
    #8 0x10038d03b in chip::app::reporting::Engine::SetDirty(chip::app::AttributePathParams&) Engine.cpp:810
    #9 0x10008af5e in MatterReportingAttributeChangeCallback(unsigned short, unsigned int, unsigned int) ember-compatibility-functions.cpp:1101
    #10 0x1001a3884 in fabricListChanged() operational-credentials-server.cpp:378
    #11 0x1001a410c in emberAfOperationalCredentialsClusterRemoveFabricCallback(chip::app::CommandHandler*, chip::app::ConcreteCommandPath const&, chip::app::Clusters::OperationalCredentials::Commands::RemoveFabric::DecodableType const&) operational-credentials-server.cpp:517
    #12 0x10023094c in chip::app::Clusters::OperationalCredentials::DispatchServerCommand(chip::app::CommandHandler*, chip::app::ConcreteCommandPath const&, chip::TLV::TLVReader&) IMClusterCommandHandler.cpp:1447
    #13 0x1002379d2 in chip::app::DispatchSingleClusterCommand(chip::app::ConcreteCommandPath const&, chip::TLV::TLVReader&, chip::app::CommandHandler*) IMClusterCommandHandler.cpp:2075
    #14 0x1002f8815 in chip::app::InteractionModelEngine::DispatchCommand(chip::app::CommandHandler&, chip::app::ConcreteCommandPath const&, chip::TLV::TLVReader&) InteractionModelEngine.cpp:1025
    #15 0x1002cb7c2 in chip::app::CommandHandler::ProcessCommandDataIB(chip::app::CommandDataIB::Parser&) CommandHandler.cpp:318
    #16 0x1002c772d in chip::app::CommandHandler::ProcessInvokeRequest(chip::System::PacketBufferHandle&&, bool) CommandHandler.cpp:150
    #17 0x1002c611b in chip::app::CommandHandler::OnInvokeCommandRequest(chip::Messaging::ExchangeContext*, chip::PayloadHeader const&, chip::System::PacketBufferHandle&&, bool) CommandHandler.cpp:84
    #18 0x1002f092e in chip::app::InteractionModelEngine::OnInvokeCommandRequest(chip::Messaging::ExchangeContext*, chip::PayloadHeader const&, chip::System::PacketBufferHandle&&, bool, chip::Protocols::InteractionModel::Status&) InteractionModelEngine.cpp:277
    #19 0x1002f461a in chip::app::InteractionModelEngine::OnMessageReceived(chip::Messaging::ExchangeContext*, chip::PayloadHeader const&, chip::System::PacketBufferHandle&&) InteractionModelEngine.cpp:518
    #20 0x1002f4d27 in non-virtual thunk to chip::app::InteractionModelEngine::OnMessageReceived(chip::Messaging::ExchangeContext*, chip::PayloadHeader const&, chip::System::PacketBufferHandle&&) InteractionModelEngine.cpp
    #21 0x100563c4c in chip::Messaging::ExchangeContext::HandleMessage(unsigned int, chip::PayloadHeader const&, chip::BitFlags<chip::Messaging::MessageFlagValues, unsigned int>, chip::System::PacketBufferHandle&&) ExchangeContext.cpp:492
    #22 0x10056c0bb in chip::Messaging::ExchangeManager::OnMessageReceived(chip::PacketHeader const&, chip::PayloadHeader const&, chip::SessionHandle const&, chip::SessionMessageDelegate::DuplicateMessage, chip::System::PacketBufferHandle&&) ExchangeMgr.cpp:320
[... cut ...]
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
```

</details>

@kpschoedel anything jumping out at you there?

bzbarsky-apple, 2022-07-13:
Actually, this is in fact a Darwin-specific problem.  I'm a little surprised we are not hitting in CI all the time; digging into why.

`LayerImplSelect::ScheduleWork` does the following:

1. Cancel the existing timer for the completion function and appState.  This is actually pretty questionable, in case the caller in fact meant to schedule the work twice...  This destroys the timer.
2. Create a timer and add it to the timer list.
3. On Darwin only, schedule a special "call HandleTimerComplete on this timer" bit, that does not get canceled if the timer is later canceled.

So if we `ScheduleWork(func, closure)` twice with the same arguments, then on Darwin we end up in a situation where the timer from the first call is destroyed by the second calle, and then eventually we have a use-after-free.

It looks like this code dates back to #7405 @vivien-apple

bzbarsky-apple, 2022-07-13:
And yes, if I just comment out the `!intersectsInterestPath` check in `reporting::Engine::SetDirty`, I get the ASAN failure above on the above-linked SHA when running the CI "Remove fabric" test.

closed 2022-07-14 by bzbarsky-apple