project-chip/connectedhomeip #20696: [SVE] TC-SU-2.5 ApplyUpdateRequest is still sent

pankore, 2022-07-13:
#### Problem
- TC-SU-2.5, step 1
- Action field is set to "Proceed"
- DelayedActionTime is set to 0
- DUT still sends ApplyUpdateRequest after transfer of image before rebooting

[tc-su-2.5-part1-otap.txt](https://github.com/project-chip/connectedhomeip/files/9105492/tc-su-2.5-part1-otap.txt)

selissia, 2022-07-19:
The attached log shows a single ApplyUpdateRequest received by the provider. This is correct behavior. 

The test step states: "Once the update is finished, verify the software version from the Basic Information cluster on the DUT to match the version downloaded for the software update. Verify on the OTA-P/TH that there is no **other** ApplyUpdateRequest from the DUT."  -- the implication being that we have one instance of ApplyUpdateRequest command as a part of the software update flow but no others. The log shows exactly that. 

This is working as expected.

woody-apple, 2022-07-19:
Cert Blocker Review: @pankore we believe this behavior is correct. Please re-open if necessary.

closed 2022-07-19 by woody-apple