project-chip/connectedhomeip #20699: Add Multiple OTA Unsolicited Message Handlers per Protocol

hubTab, 2022-07-13:
#### Problem

The Unsolicited Message Handler for Protocols framework only allows for one session at a time.

BdxOtaSender/TransferFacilitator and its supporting TransferSession have been designed and implemented to only support one transfer at a time:
- Exchange Manager implements a single call for registering an unsolicited message handler for a given protocol identifier that would be invoked for ALL messages of the given protocol (BDX in this case) via [RegisterUnsolicitedMessageHandlerForProtocol](https://github.com/hubTab/connectedhomeip/blob/1e27d92288d05b443285958525b779fe0d78cafc/src/protocols/bdx/BdxTransferSession.cpp#L174)
- TransferSession::WaitForTransfer [safeguards and enforces](https://github.com/hubTab/connectedhomeip/blob/1e27d92288d05b443285958525b779fe0d78cafc/src/protocols/bdx/BdxTransferSession.cpp#L174) the transfer state via its mState

While this behavior is appropriate for OTA Requestors, it is not enough for OTA Provider, which will need to be able to service multiple OTA Requestors at a time.

 #### Proposed Solution

Implementing multi-Bdx Transfer Session support, by extending the OTA framework to allow for handling multiple unsolicited message handlers per protocol.

bzbarsky-apple, 2022-07-14:
Is a single BDX transfer done over a single exchange?

If so, it seems like we'd still have the one unsolicited message handler, but it would create per-exchange delegates as needed....

hubTab, 2022-07-14:
The current OTA implementation allows for registering one unsolicited message handler per protocol, Bdx included.
An OTA Provider would need to be able to service multiple OTA Requestors concurrently, and the mechanism for enabling it to do so could start with redesigning the registration for unsolicited  message handlers.

closed 2023-01-20 by stale[bot]