project-chip/connectedhomeip #20713: [SVE][TC-LVL-2.2]Step 2b does not write DIFFERENT value

Kxuan, 2022-07-14:
#### Problem
The step 2b said "TH writes the OnLevel attribute on the DUT with a different value than that read in 2a".
But the test script only set 254 to the device, regardless the old value is 254 or not.
If the default value of OnLevel is 254, step 2c fails because it matches constrains "notValue: OnLevelValue".

 #### Proposed Solution
It should have another value, such as 253. If the default value on device is 254, use 253 in step 2b.

bzbarsky-apple, 2022-07-14:
@cjandhyala @manjunath-grl

manjunath-grl, 2022-07-15:
@bzbarsky-apple @Kxuan fixed in [PR](https://github.com/project-chip/connectedhomeip/pull/20821) please verify it and close the ticket

bzbarsky-apple, 2022-07-15:
@manjunath-grl Commented in that PR (though it took me a while to find the changes purporting to fix this in all the other noise; please create targeted PRs if you expect people to be able to review them).

I don't think that PR fixes this issue....

manjunath-grl, 2022-07-15:
@bzbarsky-apple sorry for the inconvinence from now on i will make dedicated PR for the issues , PR ([20821](https://github.com/project-chip/connectedhomeip/pull/20821)) is rebased and i modified [TC-LVL-2.2](https://github.com/project-chip/connectedhomeip/blob/53415c2ad4cfa98f51128209b5060427ac2dbc71/src/app/tests/suites/certification/Test_TC_LVL_2_2.yaml)

cjandhyala, 2022-07-15:
@manjunath-grl if the default value is the same as the config value, this TC never catches the failure.  you need to have extra chk that, read the value from DUT, and make sure it does not match the config value that you will write in the next step. if they are the same, then you increment that value by 1 of the config value or write a different value. Then when you read next time, make sure the value you wrote earlier is preserved by the DUT.

manjunath-grl, 2022-07-16:
@cjandhyala We currently do not have support for conditonal statements in the YAML framework, the one issue with incrementing the value by 1 is if the values is already the max configurable value it will cause an error when we increment it, By using the configuration section at least we can rerun the test case with a different values if the values are observed to be same. Let me know what you think.

franck-apple, 2022-07-18:
removing the 'request cert blocker' label, as it's a script issue.

closed 2022-07-19 by woody-apple

franck-apple, 2022-07-19:
test script repo: https://github.com/CHIP-Specifications/chip-test-scripts