project-chip/connectedhomeip #20714: [SVE][TC-LVL-3.1] Test Step 16 shoule have same PICS with Step 14

Kxuan, 2022-07-14:
**Problem**
Step 14 ("sends a Move to level command") send MoveToLevel command only if LVL.S.A0010 (OnOffTransactionTime) is supported.
If the DUT does not support LVL.S.A0010, MoveToLevel will not be invoked on DUT. So the current level are remains unchagned.

The Step 16 read current level in this condition is invalid, and should also be skipped as same as Step 14.

https://github.com/project-chip/connectedhomeip/blob/78dc599df930c05aacb49502975a214539dacee2/src/app/tests/suites/certification/Test_TC_LVL_3_1.yaml#L177

manjunath-grl, 2022-07-14:
@bzbarsky-apple TC-LVL-3.1 is fixed in [PR](https://github.com/project-chip/connectedhomeip/pull/20709) please verify it and close the ticket

Kxuan, 2022-07-16:
Thanks.
will retest on Monday.

Kxuan, 2022-07-18:
Verified. Step skipped as expected.

closed 2022-07-18 by Kxuan