project-chip/connectedhomeip #20716: [SVE][TC-DGTHREAD-2.1] Test Step 1: read configured Channel attribute value, expected value should NOT be null

yeccc, 2022-07-14:
#### Problem
<what's wrong or missing, please include any applicable:
 * expected behavior
 * actual behavior
 * steps to reproduce
 * system configuration
 * screenshots, images, logs, etc.
 >
 ***** Test Failure: channel expected to be null but isn't
BUT spec defined:
 A value of null SHALL indicate that the Thread interface is not currently configured or operational.

 #### Proposed Solution
<suggested fix, suggested enhancement>

bzbarsky-apple, 2022-07-14:
@manjunath-grl @cjandhyala

bzbarsky-apple, 2022-07-15:
I don't have a way to verify that does not involve carefully reading that whole PR.

@yeccc

manjunath-grl, 2022-07-16:
@bzbarsky-apple @yeccc please verify the test [TC-DGTHREAD-2.1](https://github.com/project-chip/connectedhomeip/blob/bd077fe03193899563ea86c63e9ec18b6da32485/src/app/tests/suites/certification/Test_TC_DGTHREAD_2_1.yaml#L31) and close the ticket

closed 2022-07-19 by franck-apple